zeevy/grblcontroller #22: Any plan support ESP8266 Websocket connection?

radiolee, 2018-01-07:
https://github.com/LaserWeb/LaserWeb3-ESP8266
https://github.com/arkypita/ESP8266-SerialWebSocket

esp8266 simple module esp-01 is cheaper than HC-05 and may be faster.

zeevy, 2018-01-07:
Looks interesting, i never had chance to work with ESP8266 module. 
With my current work schedule its not possible for me to spend time on this. I will consider this as feature request.

radiolee, 2018-01-07:
@zeevy Thanks. esp8266 with websocket fw, now have pc project like lasergrbl, web server with nodejs like lwserver, project laserweb/cncweb, grblweb .. but without an android client project.

zeevy, 2018-01-27:
Ordered couple of ESP8266 from Aliexpress. Waiting for delivery.

radiolee, 2018-01-28:
Sounds great!

radiolee, 2018-02-01:
As Laserweb4 do, controller software support websocket and socket(telnet) can send gcode via wifi network, we can forget the usb cable:grinning:

radiolee, 2018-02-05:
@zeevy I was tested some days and find out that tcp socket transport by esp-link is more stable and fast.
Please refer my tests here https://github.com/LaserWeb/LaserWeb4/issues/454

zeevy, 2018-02-05:
@radiolee thanks for sharing the information, i am about to start process of supporting the ESP8266 for this application.
Just added USB support a day ago and found it is faster than the Bluetooth connection, USB connection effectively fills the Grbl buffer (both planner and serial Rx) with the same baud rate. I guess with esp8266 also have better communication speeds.

radiolee, 2018-02-05:
I have upgrade to 2.12

radiolee, 2018-02-05:
I have update to 2.1.2 today, great works!
Refer my tests, tcp socket(telnet) from esp-link https://github.com/jeelabs/esp-link is really faster, I got 71lines/sec close to a real serialport. websocket server is slow.

closed 2018-11-21 by radiolee