zeevy/grblcontroller #38: Axis Jog Buttons Direction

aatzoho, 2018-02-16:
Very Nice Interface. On my machine from where I stand the buttons and the axis direction are reversed (so from my viewpoint I push the Left/Right buttons to move the table In/Out) . Because every DIY machine is different it would be very nice to be able to define (in setup) X or Y as the Horizontal set of buttons and which button on each axis is causes Negative motion. Thanks

zeevy, 2018-02-17:
That wolud a nice feature, can you please provide more information

aatzoho, 2018-02-17:
OK, If I lay my tablet on my CNC table, the 8 buttons on the Jog Pad do not correspond to the direction that the spindle will actually move. Now if I rotate the tablet 90 degrees counter clockwise and ignore the rest of screen the buttons and spindle are now in sync. Obviously not a good solution because now the rest of the screen is sideways. I do not know how to code so what I propose my not be practical, but here is an idea of what would make the interface flexible to all variety of machines. In the setup menu you could add a line to configure the Jog Pad, it would have 4 options "Rotate the pad 0 degrees, 90 degrees, 180 degrees, 270 degrees". It would not change the look of the main screen but it would change the logic assigned to the buttons. Example: If I set the config to 90 degrees, then the logic that was called by the upper middle button would now be called by the left middle button and all of the other buttons would be shifted accordingly.
I could imagine this to be a lot of coding on your part, If the logic for the buttons is currently coded to the button you would have to change it to variables and then call the variables from the buttons. So I am not thinking that it is easy on your part, but the result would certainly make the end product much more adaptable to the hobby CNC builder. Thanks for your time.

zeevy, 2018-02-18:
Its a nice idea to add such a feature, but it makes you machine non compatible for other GUI.
It seems you have interchanged X and Y stepper wiring. (may be due to your custom requirement).
The current Jog pad assumes machine set up done in Cartesian(XYZ) coordinate frame.
I will try to add this configurable option in future builds.

You can also use custom buttons with labels like "X+" "X-" etc.. only limitation is it will be with fixed feed rate and step unless you change it in settings.

You can find more info on machine setup here
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#quick-guide-to-setting-up-your-machine-for-the-first-time
http://linuxcnc.org/docs/html/user/user-concepts.html#_machine_configurations

zeevy, 2018-03-02:
This is implemented in version 2.1.3

closed 2018-03-02 by zeevy