zeevy/grblcontroller #64: use with OTG

wagnerteixeira, 2018-04-16:
I can not use it with OTG, the phone has support, I have another application where I use it.
when I connect the USB OTG cable, the other application asks if I want to initialize it, so I thought it might be doing some kind of blocking and I uninstalled it, still still unsuccessful.

I changed the default connection to USB OTG and it still did not work, I open and close the app and it does not connect to the grbl ... I'm using an arduino UNO R3

zeevy, 2018-04-17:
Only few serial chips are supported.
CP210X, CDC, FTDI, PL2303, CH34X, CP2130. 
What is the serial chip on the uno board?

zeevy, 2018-04-17:
Also what is the android version?

zeevy, 2018-04-17:
If you are using app version 2.1.5. Then there is a bug in that version, which prevents the USB connection

wagnerteixeira, 2018-04-17:
my arduino is like this

http://static.cactus.io/img/arduino/arduino-uno-rev3-function-diagram.jpg

zeevy, 2018-04-17:
It should work with atmega 16u2, it will pop up a dialog as soon as you connects the device. I am not sure what will be the problem may be the firmware on 16u2 chip is old. I will check and let you know

wagnerteixeira, 2018-04-17:
I just tested with another arduino, with different chip and it worked! a dialog box opened for the grbl controller. it may have some configuration on my phone, because it has another application that supports the OTG, I uninstalled it, but when I had it and I connected the arduino with the chip atmega 16u2 would open a dialog for that same application .... android have this arduino in this application..something like this

zeevy, 2018-04-17:
May be you need to update the firmware on the 16u2 chip. This application does not have drivers for the firmware on 16u2

wagnerteixeira, 2018-04-18:
I updated the 16u2 (I did not even know it was possible) and there it is! it worked!

Thank you very much for the help, thank you very much!

Congratulations for the app, it's very good, great ux / ui

zeevy, 2018-04-18:
That a good new, have a good time with your machine.

closed 2018-04-18 by zeevy