zeevy/grblcontroller #68: Jog cancel problems

terjeio, 2018-05-07:
I have experienced similar problems as you with jog cancel and have modified grbl for a (IMO) more robust version that flushes input buffers prior to cancelling motion. Maybe [my solution](https://github.com/terjeio/grbl-Mega) can solve your problem too?

zeevy, 2018-05-07:
Thanks for the solution from firmware side, it would be nice if it gets included in the upcoming Grbl version or released as bug fix on existing official release, then lot more people may get benefit of it.
And i don't think it will be fixed on the current version of the grbl. 
So here i am looking for a GUI solution to the problem, and i am pretty close to it, just need more time to implement and test it in this application.

zeevy, 2018-08-01:
This issue is addressed in the latest release 2.30

closed 2018-08-01 by zeevy