zeevy/grblcontroller #73: Cant download it

saeed477, 2018-06-29:
I cant download it because of otg. Please fix it .thanks.

zeevy, 2018-06-30:
You may need to update the serial firware on the arduino board. Try with latest board with uno R3 firmware.
Chech this thread 
https://github.com/zeevy/grblcontroller/issues/64

saeed477, 2018-06-30:
Thanks for your feedback.
But i cant update my grbl controller from google play store. Because my phome (g2 mini) doesnt have otg device and play store sayed your device dosnt support this version.

saeed477, 2018-06-30:
Android 5.0.1 official

saeed477, 2018-06-30:
Old version (without otg ) works well. But play store doesnt primission to update.

zeevy, 2018-06-30:
I am not sure what problem you are facing.
If you are not able update the through pally store, then you can install the apk directly from git hub release page

saeed477, 2018-06-30:
Thanks.
Thats evry thing that I need. Sorry I'm not pro in githup and use mobile so I didnt see that. Thanks.

closed 2018-06-30 by saeed477