zeevy/grblcontroller #74: Dwell Command "G4 P0.05"

buildlog, 2018-07-29:
I noticed that this was added earlier this year and inserts dwells into the gcode. It appears to be doing it right before a G21 and G90 command on my machine. What is the reason for this?

The commit says "send dwell command if eeprom write command detected". Neither of these commands should affect EEPROM.

I don't think that Grbl should care about this, but for some reason my ESP32 port of Grbl when running over BlueTooth. I need to fix it on my end, but I was just curious.

BTW: I love GrblController and use it all the time. 5 Stars!

zeevy, 2018-07-30:
Does G4P0 causes any problem on your machine?

buildlog, 2018-07-30:
Thanks for the quick reply. It is causing a problem on my Grbl port, but the Grbl port should be able to handle the dwell. I use the dwell trick on my automatic tool changer machine. I will try to fix my firmware.

There is a G21 command in the header and the footer of my gcode. It only pauses on the second one. I will try to figure it out, but it is probably best to remove it from Grbl Controller too. some people might not like extra gcode added.

zeevy, 2018-07-30:
I just released a update to the application in play store. It still uses G4P0 if it detects any eeprom access commands.
I have removed the non eeprom access commands from the list. You can try the new version. 
I did not find any better solution other than this.

buildlog, 2018-07-30:
I think that will help. 

Thank you, 
   Bart

closed 2018-07-30 by buildlog