zeevy/grblcontroller #99: Manual movement Z axis

bourdoul, 2019-02-16:
Hi, everybody.
I have version 2.35 (free) Not working manual movement along the axis Z. With this problem I'm not alone. The developer is sending everyone here.

zeevy, 2019-02-16:
Can you provide more details.
If only z axis is not working then may be you have chosen 0 as step size. 

Please go through the documentation available in the application about section. 

If you can provide more details or images that will really help in trouble shooting

zeevy, 2019-02-16:
And FYI this is the right place to report any problems that you may face with the application.

Doing so you are actually helping others who may have similar problem and contributing to the application. This is how open source applications works

bourdoul, 2019-02-16:
Connection via Bluetooth (HC-4 module). Through other programs everything works correctly. Moving along other axes works. The settings "Jogging max step size Z" are set to step 10. I went through different options, but this parameter does not affect the performance. The Home command works correctly-movements on the Z axis are started.

zeevy, 2019-02-17:
What is the grbl firmware version?

bourdoul, 2019-02-17:
1.1f

bourdoul, 2019-02-17:
Arduino Nano + CNC shield Board. And if I enter g-code manually (G0 Z1 or G0 Z-1), then the movements are normal.

zeevy, 2019-02-17:
Can you please share the screenshot of console window in the applicatipn.

Also axis move buttons uses jog commands not gcode commands. 

In recent version application has separate jog step bar for z axis, may be you forgot to adjust that. 

Also please share the console output appears after pressing the jog button. 

If your grbl is not custom version and it is supported by app then it will print the grbl version, settings as soon as it  onnects

bourdoul, 2019-02-17:
[image]
[image]
[image]
[image]
[image]

bourdoul, 2019-02-17:
Understood! Thanks for your help. The manual does not say anything how to call the menu "Jogging step and feed".  The menu is called by briefly pressing the line "Step Size|JogFeed"

zeevy, 2019-02-17:
Ok i got the problem
You are adjusting the max size of the step bar not choosing the step value. 

You can choose the step value in jogiing control panel it self. Click on the text "Step Size | Jog Feed" text that is just above the jog buttons. You can get a dialog to adjust z step size, xy step size and jog feed

zeevy, 2019-02-17:
Yes you are right, manual is not upto date and it is not fully documented. I will look into that when i gets time

bourdoul, 2019-02-17:
Thank you for your prompt assistance!

zeevy, 2019-02-17:
Error 2 that you are getting i am not sure about that. May some noice is picked up and interfacing the serial comunication

bourdoul, 2019-02-17:
Another question. Is it possible to make the screen not go off when the application is running?

zeevy, 2019-02-17:
Its not a good idea if will drain the battery. Application will work even screen locked.
If you need screen to be on during the job then you can do that in your phone display settings.

closed 2019-02-18 by zeevy

zeevy, 2019-03-31:
The option of keeping the screen on is now added in settings.