zeevy/grblcontroller #100: is this open source project??

mosabalrojpany, 2019-02-22:


zeevy, 2019-02-22:
Ya source code is open to public

mosabalrojpany, 2019-02-22:
thankx for this project.
but i want to edit this project add voice recognition to send g-code by voice when i but the app into android stodio give me error build .
File google-services.json is missing. The Google Services Plugin cannot function without it. 
 Searched Location: 
C:\Users\mosab\Desktop\application almshro3\grblcontroller-master\app\src\nullnull\debug\google-services.json
C:\Users\mosab\Desktop\application almshro3\grblcontroller-master\app\src\debug\nullnull\google-services.json
C:\Users\mosab\Desktop\application almshro3\grblcontroller-master\app\src\nullnull\google-services.json
C:\Users\mosab\Desktop\application almshro3\grblcontroller-master\app\src\debug\google-services.json
C:\Users\mosab\Desktop\application almshro3\grblcontroller-master\app\src\nullnullDebug\google-services.json
C:\Users\mosab\Desktop\application almshro3\grblcontroller-master\app\google-services.json

zeevy, 2019-02-22:
You can get that file really using your Google account. Just go to firebase console and create a project. I am sorry I can't provide much information on this now as I am travelling.

It is a pretty easy process, just Google it and you can find it. 

This file is required by every android app if it uses any Google services

mosabalrojpany, 2019-02-22:
ok thankx a lot zeevy

closed 2019-02-25 by zeevy