zeevy/grblcontroller #103: fusion 360 post process

SarkazmU, 2019-04-20:
Good day. I am using Fusion 360. Post processor configuration is grbl.cps v42115. Controller GRBL firmware is up to date. But GC terminal returns "Error:20 Unsupported or invalid g-code command found in block."
Thanks in advance for help
NC is attached
[14mmGRBL_nc.txt](https://github.com/zeevy/grblcontroller/files/3100265/14mmGRBL_nc.txt)

zeevy, 2019-04-20:
Hi
I think M6 tool change is not supported by grbl hence the error.

Thanks for using the app, in future I will try to address this issue.

Have nice time with your machine

Reply to: reply@reply.github.com
Cc: subscribed@noreply.github.com
Subject: [zeevy/grblcontroller] fusion 360 post process (#103)

SarkazmU, 2019-04-21:
It is. Just tested same NC without M6 and it was all correct. Thanks again.

closed 2019-04-21 by SarkazmU

zeevy, 2019-04-21:
It just a close, no solved status in github