zeevy/grblcontroller #104: Landscape mode for the application

zeevy, 2019-04-29:
Hello all,

We need landscape mode for the app. I am planning to provide this feature in upcoming releases (time line is not fixed/estimated yet)
I am not sure about the design how it should look in landscape mode mobile and tablet.

How can you help:
1. You can propose the layout designs for the landscape mode (upload images), full details are not required. Wire frame designs or simple pen/pencil design on paper are welcome.

Every good contribution will be appreciated by a promo code for the paid app.

lalamdorababu5, 2019-04-29:
I just want to a suggestion to provide a feature of tracking tool and picture of Gcode ( diagram ). So that it is easy to trace the tool and better to customize in landscape of mobile version

zeevy, 2019-04-29:
> 
> 
> I just want to give a suggestion to provide a feature of tracking of tool and picture of Gcode ( diagram ). So that it is easy to trace the tool and better to customize in landscape of mobile version

I tried to provide this feature in app with very first initial version, few problems i faced.
1. It consumes more resources in mobile.
2. It looks very messy in small screens. (UGS which runs in desktop systems still struggles a lot in visualizer)
3. Android runs on thousands of different hardware and software versions. 
4. May complex gcode paths are not looking good in mobile.

due to above problems i have given up on this part. But still i am looking for a better way for doing this by overcoming all above issues. I still have a hope some i can figure it out.

lalamdorababu5, 2019-04-29:
Can you give me a promo code for paid version

zeevy, 2019-04-29:
Promo code is accutally for good suggestions on land scape mode, since you asked for it you can mail me for the code

lalamdorababu5, 2019-04-29:
What is mail id

zeevy, 2019-04-29:
You can get that in app about section

joanorsky, 2019-04-29:
How about adding some laser control options. The grbl laser users really would love those...

[IMG_20190429_121639](https://user-images.githubusercontent.com/3766093/56892939-122c7000-6a79-11e9-8a99-ec38c529d7e2.jpg)
]

lalamdorababu5, 2019-04-29:
Yep.. Good idea

lalamdorababu5, 2019-04-29:
I am not going to take promo code for free. You gave me for free but I also
want to put my efforts. If you like or want to change please let me know

joanorsky, 2019-04-29:
Also adding a cut frame preview would be nice for material position. The preview could just be a rectangular frame where the XoY axis would run without ZZ axis movement... Or a more complex preview where the gcode outline would be traced on a XoY plane also without ZZ axis movement. This would help a lot for material placement. I say this because this feature is also present on professional control softwares and is a great help for the operator..

lalamdorababu5, 2019-04-29:
Yes I said it in the first message to you for adding frame but I told in
another form like a tool travelling. But 2D ( taking X&Y axis) is good it
is not necessary to view in 3D. Add the Gcode preview in the form of
diagram in one colour and tool point in other colour so it is
easily visible for operators. Another suggestion let me build the app for
3D printers. I am working on it

zeevy, 2019-04-29:
> 
> 
> How about adding some laser control options. The grbl laser users really would love those...
> 
> [IMG_20190429_121639](https://user-images.githubusercontent.com/3766093/56892939-122c7000-6a79-11e9-8a99-ec38c529d7e2.jpg)

I am not looking into exclusive support of laser machines. (actually i don't have a laser machine or 3d printer). The current version does already supports laser mode and works happy in that mode, few users also reported that they are quite happy with the current app on their laser machines.

My current goal is to redraw the appearance of the app  in landscape mode by still providing all the existing features (for now i have no intention to add any new features like laser, visualizer)

It is just as simple as how current app should look in landscape mode with all existing elements rearranged.

zeevy, 2019-04-29:
@joanorsky  as show in the design, i think its good to show large E-Stop button in landscape mode.

lalamdorababu5, 2019-04-29:
Good idea

rajivtctech, 2019-04-29:
Tool path visualization is waste of computational resources and provides absolutely no value of any kind.

Provide a lathe mode, where your controller sw can map diameter mode lathe code to the X and Z axes. You will suddenly land up in the league of grblgru,cwbich is the only grbl controller currently that can do that.

The ability to run lathes even if without inverse time functions or feed-per-rev functions like threading, will add enormous value to the grbl machinist...

zeevy, 2019-04-29:
@rajivtctech I agree with you, considering the cost (in terms of cpu, memory) and limitations visualization does not worth.

Lathe mode, unfortunately my knowledge on lathe (or even professional 3 axis cnc machines) is very limited, i would love to have this features implemented in the app and i hope that will happen one day with good support from persons like you, who has great knowledge on industrial cnc machines.

My current major road map for this app is the landscape mode and android things support.

georgianneacsu, 2019-04-29:
You could add a slider to change the arrows position from right handed to left handed. That could be realy useful. Rest of the buttons arranged on left side or right side, depending on the slider position.

zeevy, 2019-04-30:
> 
> 
> You could add a slider to change the arrows position from right handed to left handed. That could be realy useful. Rest of the buttons arranged on left side or right side, depending on the slider position.

Thanks for the tip, i will look into this

joanorsky, 2019-04-30:
A jog slider with dinamic intensity increase would be good (up to a configurable maximum I.e.). I agree with that..

I am more of a programmer than of a designer.. but here goes another sketch..
[image]

zeevy, 2019-04-30:
@joanorsky so in this design jog control buttons and console is always visible. 
In landscape mode we will large horizontal space than vertical space. So designs should use less vertical space.

joanorsky, 2019-04-30:
That's just a general layout.. not meant to be exactly like that. Sizes.. dimensions.. placements are not meant to be as final.. it's just an overall idea..

For instance.. the lightburn grbl software uses the jogs inside a tab and not always visible.

Besides that not all tablets have the same dimensions and the DPIs do vary modifying the final dimensions and resolutions. This is specially important because a dynamic jog will need space or else it will be hard to use as any small movement may represent a rapid increase on feedrate.. so they do need to be big if this approach is to be followed.

This is actually a little complicated (I feel your pain as a developer) as a 10" tablet will be different as of a 7" table in a usability sense..

bourdoul, 2019-04-30:
Hello to all. 
[image]
[image]

I like the interface of the application as it is now. But if it is altered into landscape mode, then I imagine it this way.

joanorsky, 2019-04-30:
It looks good to me.. but as a laser user.. eheheh.. a test fire option/frame preview would be so nice... (Yeah I know... I know...)

zeevy, 2019-04-30:
@bourdoul  i like the designs, i wanted to alter the layout just look like that. 
But instead of big for choose button we can place the console logger there, so that machine position and console both are visible in all screens

bourdoul, 2019-04-30:
Yes, it would be nice to place a logger on each screen, but I proceeded from the condition that the overall design should remain unchanged. Personally, I initially liked the ascetic design - everything is very accessible and understandable.

The main task is simply to turn into landscape mode?

zeevy, 2019-05-01:
@bourdoul yes the main task is simple land scape mode with existing elements.

zeevy, 2019-05-01:
Hi all,
how about this suggested by one of the user through email (veeramani) (all major function in single screen)

[image]

joanorsky, 2019-05-01:
Too me it looks good... but won't the smaller buttons be too small when the user's screen be of 7 inches or less?

I think you should see the statistics on Google play regarding I which equipments the application is mostly installed and design the landscape version considering the majority of the screen sizes currently on use..

zeevy, 2019-05-02:
Checked the design in 7inch tab seems to be ok.
[image]

It is more looking like mach3 UI, I think if this layout is used then look and feel should be like present app

bofred, 2019-05-02:
Great improvement.
I'm a beginner and has just started to use the app with my CNC.
--> My problem is when using the jogging feature. When holding down and feed in one direction the screen starts to flipp around and the cut gets crocked. (Xiaomi redmi note 4)
I would like to turn off the sliding of the screen. 

Otherwise great app 👍

zeevy, 2019-05-02:
@bofred I guss you are using usb connection, if yes than its the problem with electric earth connection where touch may not work properly. This is not at all app issue, app is pretty stable now.

For jogging you need use step size option provided. 
Bigger buttons are available for tablet. For phone i have to consider the different sizes, present size is optimized for most of mobile screens

bofred, 2019-05-02:
Thank you for fast feedback.

The grounding issues is indeed a problem with the frequency converter. To turn off the sliding off the screen could be a workaround. Especially when pressing down button longer times. I think its fast enough to press the buttons to move between pages. 

Have not tried Bluetooth yet. I can imagine that it would be different kind of problems with interference. 

I use the three step size buttons and I like it but a number field for long moves would be great. Especially when using the CNC as a measurement machine.

zeevy, 2019-05-02:
@bofred Current version of app allows a max step size of 500 in settings, it might be use full for you

DirtyEngineer, 2019-05-02:
Great to see that a landscape version is in the works! My one request is that a "dark mode" is added, even if it is only available in the paid version. Thanks for a wonderful app.

zeevy, 2019-05-02:
@DirtyEngineer application theming is in my to do list, where the end user can chose the colors of the application as per their taste.

mkaluza, 2019-05-06:
I could definitely use a preview window with the currently loaded script in it with highlighted lines being currently executed and an option to single step the script. Is called single block mode on CNC machines and is an extremely helpful testing/debugging feature.

And if not a window with entire script in it, then at least a preview of the next line to be executed (in single block mode) and it's line number in the file.

Lester444, 2019-11-04:
I am eagerly awaiting the release of the landscape version.

I use this now on a Xiaomi 5A to control a 5-axis flying probe tester (still in the works) and would like to have homing for all 5 axes...... but I think you have already considered that because I see axes A & B on the new layout. IMHO, the new layout looks great.

On the present version (3 axes), I am able to stream G-codes with commands for axes A & B without any hickup, Bluetooth connection via HC-05 115200 baud, Arduino MEGA clone.

If you don't mind, bigger control buttons (positioning buttons) would be great for my stubby fingers & poor eyesight. I don't really need the G-code preview. IMHO, E-stop button should not be implemented via SW. 

Fantastic software. Thank you Zeevy. I will purchase the landscape version once released.

closed 2020-01-01 by zeevy

Lester444, 2020-01-10:
I saw that this issue was closed, but I did not find a new version release.

Should I wait some more (maybe you are polishing some small details before release) or is this version no longer in the works?

Landscape version would really be nice....... :)

zeevy, 2020-01-10:
This feature is dropped as I find no time for implementing this.