zeevy/grblcontroller #106: Support request for manual tool change (fusion360)

bofred, 2019-05-02:
Hi, Im new to this and need some help to make the machine stop when a tool change is needed.

In the description of this program it says G43.1 will do the manual tool change but I don't know how to make fusion set the G43.1 or manuallu set it.

I use GRBL 1.1 and grblcontroller to handle a 3 axis CNC router.
I make the gcode with fusion 360 and the post configuration grbl/grbl.

Thank you in advance.

zeevy, 2019-05-02:
Changing a tool in the middle of a job is not possible as of now. However you can do tool change if you have exported multiple gcode files one per the tool.
You need to have probing setup. The procedure is as follows.

1. Do the first job with the first tool. (do not remove the first tool)
2. After finishing the job go to the probe location (you can use jogging)
3. Do a normal probe. (press "Probe" button in probing tab)
4. Now change the tool.
5. Press the button  "Probe + TLO"  (this will do probing and sets the TLO)
6. Jog to WCO zero location and verify that TLO is set correctly
7. Select the second tool path file.
8. Start the file streaming.

bofred, 2019-05-02:
Thank you for the extremely rapid response!

I was thinking of this workaround at first but I imagined that the functionality was implemented when I looked at the intro page. Is it the grbl who does not have the support or can it be fixed in the app later on. 

Thank you once again.

closed 2019-05-02 by bofred

zeevy, 2019-05-02:
M6 command not event supported by grbl due to various good and its won reasons.
It can be done on GUI side with user specified macro. (this is not currently available in app)