zeevy/grblcontroller #107: Scroll in the terminal window

Lenrockus, 2019-05-21:
It seems that scroll , doesn't work in the terminal window. V1.25 grbl controller+ neither v1.24

zeevy, 2019-05-22:
Thank you for reporting,
I will check, can you tell me which version of android your are using

Lenrockus, 2019-05-22:
5.02 and 7.0

Lenrockus, 2019-05-24:
same problem v1.26

zeevy, 2019-05-24:
I am unable to reproduce this problem.
Tested in Android 4.4, 6, 8.1, 9
I don't have device with 7.0 or 5.02. 
I do have android 5.1 phone, i will check with that

zeevy, 2019-05-24:
Does the scroll working for you in free version?

Lenrockus, 2019-05-24:
Yes, it works free version. It also works on paid version 1.23

Lenrockus, 2019-05-24:
[https://youtu.be/bHV9P1n6624](url)
here is video of how it looks like

zeevy, 2019-05-24:
Yes that looks like an issue with small screen mobiles.
Thanks for the video. I think I know where to fix it. 

Can you try by decreasing the display size if supported by your phone.

Lenrockus, 2019-05-25:
Unfortunately my phones doesn't support display resizing.

Lenrockus, 2019-05-25:
Sorry, on android 7 it is possible to decrease  display size and the scroll is working. On normal screen size it doesn't.

zeevy, 2019-05-26:
OK, i will fix the issue soon

zeevy, 2019-06-09:
this is fixed in free version of the application, you will get the update notification soon.
Please check and confirm the same

zeevy, 2019-06-11:
This issue is now fixed in both paid and free version

closed 2019-06-12 by zeevy