zeevy/grblcontroller #108: How to force unlock

beckmx, 2019-05-26:
Hello, i was testing the app a little bit and world like a charm, however sometimes i find It difficult yo make It Connect propperly, I know that It is connected because in other apps It responds but in the controller doesnt activate the buttons, os there any way to force unlock the gui?

zeevy, 2019-05-27:
There is no option to force unlock the buttons.
Buttons gets activated once grbl firmware is found. 

Few buttons in the app gets activated only in machine ideal state.

beckmx, 2019-05-27:
The thing is, my bluetooth adapter is only connected to tx and rx, so even if I manually reset the arduino and it sends the GRBL 1.1 message it doesnt unlocks

zeevy, 2019-05-27:
I think I do not understand what you are trying to say.
Can share the screenshots? 

Also most of the buttons will not work in alarm mode

beckmx, 2019-05-27:
i will record a video

beckmx, 2019-05-28:
I was trying to do it myself downloading the code but I get this issue:

`Process: in.co.gorest.grblcontroller, PID: 13062
    java.lang.RuntimeException: Unable to start activity ComponentInfo{in.co.gorest.grblcontroller/in.co.gorest.grblcontroller.BluetoothConnectionActivity}: g.b.a.g: Subscriber class in.co.gorest.grblcontroller.BluetoothConnectionActivity and its super classes have no public methods with the @Subscribe annotation`

Do you know what could be missing?

zeevy, 2019-05-28:
The error says, event bus is registered, but class does not have any subscribers.
Have you removed any methods in that class?

zeevy, 2019-05-28:
If you just want to force enable the buttons. 
there are two ways
1. remove the enable attribute of every button in all xml layouts.
2. the method "getState" in https://github.com/zeevy/grblcontroller/blob/master/app/src/main/java/in/co/gorest/grblcontroller/listeners/MachineStatusListener.java#L103 should always return "Constants.MACHINE_STATUS_IDLE"

beckmx, 2019-05-28:
Hello, tes I noticed that línea but i didnt novedades anything else :( , i read some posts that maybe is proguard

zeevy, 2019-05-28:
Ya you are right. Its a proguard issue.
You can disable it in
https://github.com/zeevy/grblcontroller/blob/master/app/build.gradle

minifyEnabled false

beckmx, 2019-05-28:
sorry for my bad comment from the autocorrect lol, so it finally compiled without the lock feature, I had to add these lines for Mac with java 1.8:

`android {
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
} `

and 
`minifyEnabled false`
only for debug, after that it started to work :), thank you for the support, you are awesome on this sharing

closed 2019-05-28 by beckmx