zeevy/grblcontroller #110: voice writing machine

mosabalrojpany, 2019-06-08:
hello zeeve.
i want to thank you for your support anyway and we complete our project about voice writing machine and we edit android grpl controll and add voice extension there a video in youtube about our project this is the url .
https://youtu.be/O8wGxSXQVrE

zeevy, 2019-06-08:
Wow! congratulations for your hard work.
Its really nice.
How many words it will take at a time?

mosabalrojpany, 2019-06-08:
This depends on your design of the g-code file where the file comes and writes after pronouncing the filename by the assets file

zeevy, 2019-06-08:
I mean when we speak, do we need to speak letter by letter or words

mosabalrojpany, 2019-06-08:
You can speak letters and words

zeevy, 2019-06-09:
ok

closed 2019-06-09 by zeevy

Janaki-Ramana, 2021-08-19:
Hii bro,i have done a Bluetooth cnc writing machine.can you pls provide me the voice controlled grbl controller app.pls so i can make it as voice controlled with your help.hoping that you will send the app link here or to mail.pls this is a request bro