zeevy/grblcontroller #111: Application crash when invoking probe or terminal screens

mkraska, 2019-06-10:
Application crashes when invoking probe or terminal screen, no matter if a device is connected or not

Jogging works, custom commands via buttons also work.
Invoking the file screen also works (streaming not tested yet)

version 2.40 on Android 9, Motorola G6

zeevy, 2019-06-11:
Thanks for reporting.
I fixed the issue. Can you please check with the recent version 2.43

mkraska, 2019-06-11:
Thanks for fixing. Works now. Great work.

closed 2019-06-11 by mkraska

zeevy, 2019-06-11:
Ok, thanks for reporting. 
Do you have "Grbl Controller +", if not please give me a mail i will provide promo code.

mkraska, 2019-06-11:
> | mute the thread ] .

Hello, 

Thank you, a promo code would be nice! I'd give the + version a try. 
BTW, your hardware setup pages helped me a lot to get the configuration of the bluetooth module right. I adjusted the settings with a small M5Stack UIFlow program, because I didn't have an Arduino at hand. 
If you want, I can provide the contents for a wiki page (source, screenshot, hardware photo and some explanations). As of now, the tool is rather trivial and not very user friendly. 

Best regards, Martin Kraska 

Best regards, Martin Kraska 

-----

zeevy, 2019-06-11:
Sure you can add those hardware setup steps to wiki.
You can clone the this repo and add/change the wiki pages in docs folder (you need little knowledge of html css) and create a pull request.

If you don't know much of html/css then provide me the content i will update at my end.

You can mail me from your email address so that i can share the promo code. (my email id can be found in about section of the application)

mkraska, 2019-06-11:
My mail is [ mailto:kraska@th-brandenburg.de | kraska@th-brandenburg.de ] 
I was not aware that the system hides the sender email. 

I'll see how I can add a wiki page, I usually use Markdown in Github. 

Best regards, Martin Kraska 

-----

zeevy, 2019-06-11:
Ya Github stopped personal communication between user some time ago.
App wiki pages are in html format and markdown is not supported.
I have sent the promo code to your email.