zeevy/grblcontroller #112: Reasonable defaults for quick step buttons

mkraska, 2019-06-11:
1. Currently, the defaults for the quick step buttons in the step/feed adjustment widget, are all zero. They might be set to 0.1, 0.3 and 1 or whatever might be useful.
2. Also, I would recommend to increase the jog speed with increasing duration of keypress. I.e. you would start with a very low speed and get increasing speed as longer as you press (a la Grbl Overseer). Min and max speed and increment should be customizable. 
1. I recommend to join the settings of the quick buttons and the max values to a single widget? They affect the same UI element anyways.

zeevy, 2019-06-12:
Steps 1 and 3 are good idea.
Step 2 bit complicated and seems not resalable for all scenarios (may be it should be a optional feature)

zeevy, 2020-01-01:
Implemented in recent version

closed 2020-01-01 by zeevy