zeevy/grblcontroller #114: Is GRBL Controller editing $10=2 by itself?

Lester444, 2019-11-07:
I had a lot of trouble after evaluating, not while using GRBL Controller but with bCNC. At some point I had to go back to bCNC to run my CNC machine normally. I keep getting "Not Connected" and the coordinates do not update, can't zero, etc. But I know it is connected because I get a reply with $$.

It took me a lot of time & effort to get my machine working again, and would have saved me a lot of trouble if I knew beforehand that BCNC needs $10=1 (it was indeed on $10=1 before I connected via HC-05), while GRBL Controller uses $10=2 (I think, not sure).

Would be better if this $10 is edited manually, IMHO. Save some folks a lot of pain & time.

zeevy, 2019-11-07:
Yes this app needs buffer status, so it will enable $10=2
This is also said in app description in play store.
This setting should not be a problem with any other software.

May be I should set it to normal on disconnect

Lester444, 2019-11-07:
> May be I should set it to normal on disconnect >   good idea :+1:

closed 2019-11-07 by Lester444