zeevy/grblcontroller #115: I want to buy 5-axis version but I don't know how it looks like

Lester444, 2019-11-09:
Does somebody have a screenshot?

I don't have details about GRBL Controller+ version, if it will work for what I am making. Also, can I use this on my smartphone, and also on my tablet? (not at the same time)

I need jogging control for a 5-axis machine and description in GooglePlay seem to indicate that it is possible. But how does the screen look like?

zeevy, 2019-11-10:
[image]

This is how it looks like. 
It supports à additional A and B axis for jogging only. 
The status of the axes ie current position is not shown. Just two buttons.

Lester444, 2019-11-10:
I see that I can jog axes A & B, it is good enough. Now is there a way to zero those axes after jogging? Even manual input of G92 would be good enough.  Even system reset after jogging would be acceptable. I intend to run G-codes on this machine.  (It's a flying probe tester with X1-X2-Y1-Y2-Z-Z  -->  X-Y-A-B-Z-Z)

zeevy, 2019-11-10:
No option to zero them in app.
But you can any two of the four custom buttons. 
They support both short click and long click

Lester444, 2019-11-10:
> But you can use any two of the four custom buttons.> 
This great Zeevy, thank you. I think I can manage that.
Last question before I buy:  can I use this on my smartphone, and also on my tablet? (not at the same time)
Reason: I have not purchased a tablet for this machine yet. I will do so eventually, before I deliver it to the customer. I am using my smartphone right now as an interface (debugging, trials & evaluations, etc.) and I will need to be able to use the same capabilities later, on the tablet.  Or do I have to buy a license for each one?

BTW, when you release the new version with the new screen layout, do I need to purchase another license for that release?

zeevy, 2019-11-10:
Once purchased you can install the app in any number of divises. But you must use same Google account in all of those divices.

Once purchased all future updates are free. 

You can mail me so that I can share the free promo codes of the app. I have few of them. You can find the email in app about section.

Lester444, 2019-11-10:
> You can mail me so that I can share the free promo codes of the app. 

> Thank you for the offer Zeevy, but I would rather support your work. I have already purchased a license and I will be trying it out over next few days.

closed 2019-11-10 by Lester444