zeevy/grblcontroller #116: Request: Additional Setting for Minimum Step Size

Lester444, 2019-12-09:
When I slide the bar to go for a small jogging increment, i.e. 0.01mm, I always overshoot (undershoot?) and end up with 0.00mm.... which is useless for jogging. It is a struggle to get 0.01mm on the slider.

If I can set this minimum step size somewhere then it would be very helpful.

zeevy, 2019-12-09:
0.01 can be added to quick step buttons in app settings. Just have a look in jogging preferences

Lester444, 2019-12-09:
I see that I can add 0.01mm to quick step buttons. So how do I ACTUALLY USE those quick step buttons for jogging? I've spent 30 minutes playing with the jog controls but nothing there....... and I can't find directions for use in  --> https://github.com/zeevy/grblcontroller/wiki/Interface-Documentation

Also, I think it is better if the slider has an exponential response, just like the volume control knob on an amplifier where there's more resolution at the lower end of the scale.

zeevy, 2019-12-10:
There are three quick step buttons for each slider in settings (low, medium, high) .
Assign 0.01 value to the low button.
Open the slide dialog and click on the low button. 

You can also set "jogging max step size for z" to 1 then slider max value becomes 1, after that your can easily select 0.01 on slider.

Exponential scale is good idea, i will check this

zeevy, 2019-12-10:
Updated documentation is in app about section

Lester444, 2019-12-10:
got it!  :)  --> Open the slide dialog and click on the low button.

closed 2019-12-10 by Lester444