zeevy/grblcontroller #117: HMSoft HM-10 BLE support

abdset, 2020-01-14:
This is not an issue, but i just wish if you could add support for BLE Bluetooth device HM-10 whitch uses BT 4.0 mode and low energy feature. The BT speed is better than HC-05 or HC-06 but the grbl controler app can not connect to it from android (6 to 9) phones when its connecting perfectly if i use HC-xx bluetooth 2.0 devices.
The phone (android 9) has no issue to connect with HM-10 when using other apps like Serial-Bluetooth app where i could send text commands to my CNC machine such as $H, X200, .....
This shows that the devices themselfs are ok to connect but grbl controler need some improuvement to make that happend.

If this is fixed i will buy the pro version of grbl controler just because i found that it is the best app for cnc, it is just perfectly made and all my needs are in it.

Thank you so much for your efforts.

zeevy, 2020-01-14:
This is in my todo list. Not sure when it will complete.

fsvelim1, 2021-07-26:
One more vote for BT BLE support ;-)

leo1988joseph, 2022-11-10:
Does the GRBL support the BLE now?

zeevy, 2022-11-10:
No not yet

closed 2026-11-07 by zeevy