zeevy/grblcontroller #118: Остановка передачи gcode в arduino grbl

Mickle1977, 2020-03-08:
[image]
Подскажите, что может быть не так с этим кодом?
Код сформирован в ArtCam9.
Пропуск ошибок в настройках программы включен.
[image]
При включении данного режима, остановка происходит после первой команды.

zeevy, 2020-03-08:
T1M6 is not supported by grbl.
What you have enabled is single step mode. You should enable is ignore error 20
[image]

zeevy, 2020-03-08:
Also update the app to latest version

Mickle1977, 2020-03-08:
Режим игнорирования ошибки включен. Пошаговый выключен.
В обновленной версии точно будет работать все? Новая версия платная, поэтому хотелось бы проверить работоспособность на тестовой.

zeevy, 2020-03-08:
There seems to be a bug in free version, due to which the setting is not working. ( i just checked)
please use this promo code to download the paid version. (it may take time to fix this issue)

```
7FEGSHB28HNPG5HEGEYR8X4
```

Mickle1977, 2020-03-08:
Спасибо.
Убрал команду T1 и все заработало

closed 2020-03-08 by Mickle1977