zeevy/grblcontroller #120: can you make the software conect to ANY grbl version?

sinfocomp, 2020-04-12:
, y have modified my GRBL version to acomodate for my own setup and thus i have modified the connection string by appending more info
however the software doues note connect (or does not recognize the connection)

zeevy, 2020-04-13:
Major functions of the app depends on the default grbl firmware, there is no guarantee that it will work for custom ports. As mall function is very dangerous here i cant provide support for the custom ports.
If you can leave the startup string same as original grbl, then you can test it out.

sinfocomp, 2020-04-21:
Can a "use any version" check mark option be used, so we can choose to asume the risk , at least on the pay versión, (i support on play store)

zeevy, 2020-04-22:
I will check this possibility in next release

zeevy, 2020-05-24:
implemented in v2.56

sinfocomp, 2020-06-06:
zeevy, thanks for the enhancement, it now connects, however the functions do not get active, looks like the main conection routine works but i would guess the conection string gets checked somewhere else, 
as a suggestion, can a check box be used to "conect to anything" instead of asking for the init sring ?

zeevy, 2020-06-08:
May be you are having different bluetooth buadtate issue.

What is your statup string?

sinfocomp, 2020-06-08:
im using correct baudrate becouse on the terminal screen i can see it connects and send commands mannually however all other functions do not activate 
Startup string is "Grbl 1.1 MX Plasma"

sinfocomp, 2020-06-08:
most other senders just expect the "Grbl 1.1" part and ignore the rest of the string,

zeevy, 2020-06-08:
I am expecting the custom string to be like this
Mx Plasma 1.1
Version number at the end may required for current implimentation

sinfocomp, 2020-06-08:
ohhh, i see, ok i will change my string and give it a try,
Thanks!

zeevy, 2020-06-08:
Not sure, as you said before I think better if it would be a check box instead of input for the end user.

sinfocomp, 2020-06-08:
even better, why not just ignore all caracters after the "Grbl 1.1" part

sinfocomp, 2020-09-30:
any update on this ?, please help

zeevy, 2020-09-30:
Sorry, too busy.
Will check as soon as possible

closed 2026-11-07 by zeevy