zeevy/grblcontroller #121: not working

bloodkick07, 2020-04-25:
i can only connect to my bluetooth module hc05 it shows connected but its not working

zeevy, 2020-04-26:
Please  follow this guide
https://zeevy.github.io/grblcontroller/bluetooth-device-setup.html
And make sure you are using grbl 1.1

bloodkick07, 2020-05-02:
i followed the steps but it doesnt work please help me ts for my graduation
i will do anything for your help and thank you

zeevy, 2020-05-02:
Please answer these
1 which version of grbl your are using 
2. Which connection you are using bluetooth or usb. 
3. If you are using bluetooth then wich bt module you are using. 

Posting pictures of your setup will help solving the problem

bloodkick07, 2020-05-02:
I'm using hc05 Bluetooth module. Its connected with arduino and cnc shield

zeevy, 2020-05-02:
Grbl version?

Have you changed the baudrate of the HC-05 to 115200?

bloodkick07, 2020-05-04:
Grbl? Im not using grbl

bloodkick07, 2020-05-05:
aaaa you mean the grbl installed in arduino ?

bloodkick07, 2020-05-05:
i m using 1.1

bloodkick07, 2020-05-05:
im just trying to connect the app to hc05 ,but other components like motors nema23 isint connected ,,am i obliged to connect everything and try again 
??

zeevy, 2020-05-05:
For bluetooth connection just make sure you have changed buadrate of hc-05 to 115200.

bloodkick07, 2020-05-15:
I did but not working

zeevy, 2020-05-24:
> […](#)

Please verify this, app will work buttons will be enabled once you connect to grbl 1.1 with correct buadrate

bloodkick07, 2020-06-07:
there is an error notification when i try to use the buttons : machine not
IDLE

zeevy, 2020-06-08:
Home the machine before you use it

bloodkick07, 2020-06-08:
what does it mean ? home the machine

bloodkick07, 2020-06-09:
how to home  the machine

zeevy, 2020-06-10:
Click on the home icon in jog pannel.
Please go through the app documentation at least once

bloodkick07, 2020-06-10:
I have told you that icons are not working. Can't click on them

zeevy, 2020-06-10:
Please go through the documentation and set up the bluetooth connection properly

bloodkick07, 2020-06-16:
when i press the button it shows machine not idle
Even the home button

zeevy, 2020-06-16:
Please shoot a video with another phone.
Upload it to youtube and share the link

bloodkick07, 2020-06-16:
I don't have a YouTube channel

zeevy, 2020-06-16:
You don't need the youtube channel for uploading the videos.
you can use your google account.

If not youtube then you can share the video from any other ways

bloodkick07, 2020-06-21:
This is the configuration which i did

Le dim. 21 juin 2020 à 11:56, Skander Benahmed <

bloodkick07, 2020-06-21:
Here is the connection of the Bluetooth module

Le dim. 21 juin 2020 à 11:57, Skander Benahmed <

zeevy, 2020-06-21:
Have you shared any images i am unable to view any

bloodkick07, 2020-06-22:
I've send you the Bluetooth connection.have you received the video of the
app?

bloodkick07, 2020-06-22:
https://drive.google.com/file/d/12WvwXsHRJ5CbhoQwtkX_pGmLYhTv3orJ/view?usp=drivesdk

Le lun. 22 juin 2020 à 17:28, Skander Benahmed <

bloodkick07, 2020-06-22:
https://youtu.be/UdRCLoeffVk

Le lun. 22 juin 2020 à 17:32, Skander Benahmed <

zeevy, 2020-06-22:
Oh sorry bro.
You are using grbl 0.9j vrrsion which is not supported. 
App supports only grbl 1.1 version

bloodkick07, 2020-06-22:
Can you please give me a link to download grbl 1.1 version

bloodkick07, 2020-06-23:
Thank you it works
One thing
When i want to home the machine it tells me homing is enabled via setting
How to fix that?

Le lun. 22 juin 2020 à 23:55, Skander Benahmed <

zeevy, 2020-06-23:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbl-settings

From settings you can enable it. 

Finally you got it working good to know

closed 2020-12-07 by zeevy