zeevy/grblcontroller #122: file picker stucks at /storage/emulated/0

MrDiesundas, 2020-05-06:
Hi
file picker for .gcode file stucks at /storage/emulated/0.
App was enabled to access file system
Smart Phone:
Huawei Android EMUI Version 10.0.0
App Version: GRBL Controller+
Please help.
Kind Regards 
JOE
[image]

zeevy, 2020-05-06:
Its looks wired to me actually is should show like this

[image]

zeevy, 2020-05-06:
May be app has issue with android 10 version.
Can you please check in other mobile and let me know

MrDiesundas, 2020-05-06:
thanks for fast respons.
what android version and phone do you have?
kind regards
Joe

MrDiesundas, 2020-05-06:
android 7 and 8 works (tested on kids phone). could you investigate my issue? i can not downgrade my phone.
Joe

zeevy, 2020-05-06:
May be issue with android 10.
Currently I don't have android 10 device to test. 
I will check this soon.

zeevy, 2020-05-06:
Currently app runs on morethan 890 android divices with version 10, it is stange no one reported this issue

rasanta99, 2020-05-13:
I have the same problem. I'm running it on a pocophone f1 with Android 10

IceOnly, 2020-05-13:
I have the same Issue. On my tablet and phone. Tablet runs Android 7.1.2 and my phone (Huawei Mate 20 lite) runs android 10

romadhon-razzan, 2020-05-20:
I have the same Issue, on my phone samsung Android 9 (works). but on Android 10 (not works)

whins, 2020-05-24:
Issue is fixed with last update
Thanks

closed 2020-05-24 by zeevy