zeevy/grblcontroller #123: RESUME PROBLEM

sinfocomp, 2020-06-06:
Hello, i have detected the following problems and offer some advice of posssible solutions
for most CNC operatonz Z0 is the machine table so given the following Job resume process:

Application will automatically prompt for job resume if it detects your most recent job was incomplete, and you are doing the same job again.
OK

Application will suggest a number of lines to skip, you can change the number to whatever you want.
Application will parse and skips the number of lines given.
OK

Restores the feed rate, parser status etc..
OK

Decreases the rapid motion to 50%, and move the Z axis to zero position in machine coordinates.
PROBLEM: if machine zero is the machine table cutter will dive into material, or if using plasma, torch tip will be in contact with material surface.
SUGESTION: establish a safe Z setting on program so if resume is needed this Z height would be used here

Restores machine position of X, Y axis. (also restores A, B axis if applicable)
OK

Enables coolant if required, starts the spindle if laser mode is not active.
TO BE TESTED ON PLASMA

Restores the Z axis. (this is rapid move with 50% feed rate)
OK

Waits for 10 seconds before resuming streaming. (Here you have enough time to stop streaming if you noticed anything wrong or inappropriate.)
ENHANCEMENT: Dialog showing is machine ready to continue Y/N and get feedback from OP

Restores the rapid overrides to 100%. (If laser mode is enabled laser will turned on at this stage instead of in step 7.)
TO BE TESTED

Continues streaming normally
OK

Thanks

zeevy, 2020-06-08:
Z zero in machine coordinates should work, unless user has done some changes to them.

Asking the user for safe z value is good idea.

sinfocomp, 2020-06-08:
Machine Z0 is work table so this will not work on many machines and even if Z0 work coordinates are used on top of work surface you would still want safe Z height to avoyd crashing into clamps or holders

zeevy, 2020-06-21:
Then it better to ask the user to enter it with a suggested value

closed 2026-11-07 by zeevy