zeevy/grblcontroller #125: Please help compile

sinfocomp, 2020-07-06:
I have installed android studio, can some one please help me on how to compile 
Thanks

zeevy, 2020-07-07:
keep a eye on the errors thrown by the android studio, by solving them you can build this.
IDE will show you the error.

zeevy, 2020-07-07:
https://github.com/zeevy/grblcontroller/issues/101

sinfocomp, 2020-07-07:
Thanks, for the reply, can you please Point me out wich is the Main proyect file i should open to compile in  androidstudio
Thanks

zeevy, 2020-07-07:
you need to select the project folder and import it into the Studio

sinfocomp, 2020-07-07:
Ok, i did, i get a small box saying "inválid VCS root mapping"

sinfocomp, 2020-07-07:
[image]
here is the screen after some downloads and updates

sinfocomp, 2020-07-07:
updated licences and now i get following error

KotlinDslScriptsParameter(correlationId=3546988454677, scriptFiles=[]) => StandardKotlinDslScriptsModel(scripts=[], commonModel=CommonKotlinDslScriptModel(classPath=[], sourcePath=[], implicitImports=[]), dehydratedScriptModels={}) - took 0.001 secs

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.

Ams-Laser, 2022-04-16:
can any one please help me compile this ?

zeevy, 2022-04-17:
What is the problem/error you are getting?
Generally Gradle errors are self explanatory

Ams-Laser, 2022-10-11:
Can you please let me know the versión of Android studio and graddle you
use to compile
I want to have the same setup so i can compile

zeevy, 2022-10-12:
> Can you please let me know the versión of Android studio and graddle you use to compile I want to have the same setup so i can compile El dom., 17 de abril de 2022 12:32 a. m., zeevy ***@***.***> escribió:
> […](#)

https://github.com/zeevy/grblcontroller/blob/master/gradle/wrapper/gradle-wrapper.properties

closed 2023-01-16 by zeevy