zeevy/grblcontroller #126: auto levelling

PolSerg, 2020-10-27:
Hello!
Are you planning to add mesh surface measurement and autolevel for PCB milling to your program?

zeevy, 2020-10-27:
Its not in my todo list, I will check this option in future (may be not soon)

MeKeCNC, 2020-11-01:
Hi zeevy 
I want to buy your software from Google store, but I need autolevel for PCB.
When can you add mesh surface measurement and autoleveling for PCB milling to your program?
thank you

zeevy, 2020-11-02:
I can't say when it will be, it may take lot of time

MeKeCNC, 2021-03-01:
is there any progress for auto leveling?

zeevy, 2021-03-01:
No, nothing interesting at this time

damped, 2021-07-04:
I would also really love this feature. Is there any way we can help to implement it?

Maybe after reading the gcode file determine the maximum x and y movements in the file and probe with a user defined spacing in that area. This would keep the UI modifications minimal and easy to use.

mgritsan, 2022-06-03:
Any news guys?.  I really need autolevel.  Thanks