zeevy/grblcontroller #127: Jog Feed Rate Custom Setting

rlwoodjr, 2020-10-31:
I have my Grbl Controller in inches and I have customized the step sizes for XY and Z  (low, medium, and High)  However, I cannot seem to modify the jog feed rate.  The max is set at 38100 inches per minute.  My KL744 top speed is 1000 inches per minute.  

How can I adjust the maximum jo feed rate to match my machine?

Thanks,
Bob

zeevy, 2020-10-31:
There is no customization setting for this value.
Max feed rate is taken from grbl settings. 
$110 $111 $112 max value of these settings is taken as max feed rate for jog

rlwoodjr, 2020-11-01:
Yes, but grbl firmware values are set in millimeters.  Should it change if you select inches in the jog step size dialog box?  38100/25.4 =1500 inches per minute

zeevy, 2020-11-01:
Now I understand the problem. It will not change to inches.
I will address this in upcoming releases

closed 2023-01-16 by zeevy