zeevy/grblcontroller #134: Can I choose a line of processing? Can the cutter be cut off or the power cut off in the middle of the process?

txg5230, 2021-02-25:
If the power is cut off or the knife is cut off or other reasons, the machine must start carving from scratch, which is very inconvenient and a waste of time. I don't know if this controller has this function.

zeevy, 2021-02-25:
Paid version has that feature called job ressume. Works with absolute coordinates

txg5230, 2021-02-26:
How to pay, I'm from China

txg5230, 2021-02-26:
Where can I download the paid version

zeevy, 2021-02-26:
If google play store is allowed in your country, then you can find it in the play store
Grbl Controller plus

txg5230, 2021-02-27:
I can't use the software in our country. I like it very much. Is there any way to pay you RMB directly? Does your source file have the function of paid version?

zeevy, 2021-02-27:
https://play.google.com/store/apps/details?id=in.co.gorest.grblcontroller.plus

The only way you can get it is from google play store. 
This source does not contain the functions of paid version

txg5230, 2021-02-28:
China can't use Google Store, can I find someone else to download it and send it to me? Or whose account downloads who can use this application?

txg5230, 2021-02-28:
Not downloaded from my account, can I use this app

zeevy, 2021-02-28:
Sorry to say there is no other way to download the paid app. You have to use Google play store only may a VPN service will work for you.
If you are unable to pay for any reason, i can provide you the promo code. But still you need to download via google play store only

txg5230, 2021-02-28:
Downloaded in the Google Store, can this app be installed on any mobile phone?

zeevy, 2021-02-28:
Can be installed in android phones with version above 4.4 with bluetooth or usb OTG connection

txg5230, 2021-02-28:
I downloaded the software, but one mobile phone can't install it, the other mobile phone can install it, but it doesn't work properly. Click the "open file" button, and the app will close automatically. Is APP2.68mb normal?

zeevy, 2021-02-28:
Apk file won't be more than 3mb of size

zeevy, 2021-02-28:
Also you can't download the paid version like that.
For free version apk you can check the releases in this repo

txg5230, 2021-02-28:
[image]

txg5230, 2021-02-28:
[image]

txg5230, 2021-02-28:
The free version works well. Unable to find bgu

txg5230, 2021-02-28:
Can only download that person to use the APP? Or where to set it before it can be used normally

zeevy, 2021-02-28:
Sorry I am not sure what you are asking. To my knowledge grblcontroller + can be downloaded only from play store

txg5230, 2021-03-01:
[image]

txg5230, 2021-03-01:
[image]

txg5230, 2021-03-01:
Download the app from the website you gave

txg5230, 2021-03-01:
Click the file button APP to automatically close, unable to load the path. After several mobile phones have been replaced, this is the case in the test, and the operation of APP will automatically exit.

txg5230, 2021-03-01:
[image]

txg5230, 2021-03-01:
One of the mobile phones can install the APP, and the connection to Bluetooth is also successful, but all the buttons are gray

zeevy, 2021-03-01:
If your phone does not support google play store, then app may not work properly

zeevy, 2021-03-01:
> One of the mobile phones can install the APP, and the connection to Bluetooth is also successful, but all the buttons are gray

May be the bluetooth module is not configured for the buadrate of 115200. You need to change the bluetooth module buadrate to 115200

txg5230, 2021-03-01:
Google withdrew from China in 2010 and could not use Google Store. It should be for this reason that it cannot be used normally. Is there any way to solve it

zeevy, 2021-03-01:
> Google withdrew from China in 2010 and could not use Google Store. It should be for this reason that it cannot be used normally. Is there any way to solve it

Unfortunately nothing I can do in this scenario. 
May be you need to build your won with the source code

txg5230, 2021-03-01:
Thank you very much for your answer. It is very difficult to pay for the download. It is a pity that the download is still useless. This APP is the best I have ever used. I hope the free version also adds a little more features. Development

closed 2021-03-12 by zeevy