zeevy/grblcontroller #135: Support for normal USB connection. (Raspberry Pi 4 with Android 11)

ChrisAnd1998, 2021-03-09:
Currently i'm trying different solutions to build a gcode sender. I want to do this with a Raspberry Pi 4 with touchscreen. So i came across your nice GRBL Controller app and it worked fine on my phone with OTG. I decided to install android 11 on my Raspberry Pi 4 but sadly the app gives the error "Device not supported". I'm sure this has to do with the USB connection not being a OTG connection. Would it be possible? This seems to be the best solution for my project... Android is just the best for touchscreens and easy to use.
I can just use Raspberry Pi OS with UGS Platform or Classic but that's just not really usable and slow.

Maybe you have a Raspberry Pi 4 laying around to test it..
https://konstakang.com/devices/rpi4/

zeevy, 2021-03-09:
USB driver in the app supports only few chips. Have you tried bluetooth connection?

ChrisAnd1998, 2021-03-09:
Haven't tried that yet. I kinda want to avoid using bluetooth but if there is no other option I will...

ChrisAnd1998, 2021-03-09:
@zeevy Allright, something weird is going on. I have been trying to communicate with GRBL manualy with : Serial USB Terminal
https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal&hl=en_US&gl=US

When i uninstalled Serial USB Terminal i couldn't connect anymore. So the actual fix is to have Serial USB Terminal installed and have it set to the right braud rate 115200. Now it keeps working.

ChrisAnd1998, 2021-03-09:
@zeevy After multiple restarts it's just auto picking the wrong USB. 2 are connected 1 for touchscreen and the other for GRBL. Sometimes it picks GRBL (which works!) and sometimes the touchscreen. It can be fixed by having an option to manually set a device as default.

closed 2021-03-09 by ChrisAnd1998

zeevy, 2021-03-12:
> @zeevy After multiple restarts it's just auto picking the wrong USB. 2 are connected 1 for touchscreen and the other for GRBL. Sometimes it picks GRBL (which works!) and mostly the touchscreen. It can be fixed by having an option to manually set a device as default.

Manual selection can be good option in case of multiple USB devices (never faced this issue in mobile/tablets as they have only one usb port)

ChrisAnd1998, 2021-03-12:
@zeevy Would be very nice if you can add it! It will give people so much more possibilities for their projects. I could have used an old tablet but wanted something replaceable and a Raspberry gives that. I have been trying to temporally fix it..

This is what i came up with on **GrblUsbSerialService.java** line **296**:

First i tried hardcoding the ProductID and VendorID:
`if (deviceVID == 0x2341 && devicePID == 0x0042)  {`
Which works fine for me.

Now i retrieve the ManufacturerName and if it contains **Arduino** then use that..
This works for both my UNO's and MEGA's from different brand's.
Maybe this is a simple solution to keep auto selection. Not sure if every Arduino contains **Arduino** in the ManufacturerName
 
```
String deviceNAME = device.getManufacturerName();

if (deviceNAME.indexOf("Arduino") > -1) {
```