zeevy/grblcontroller #136: [Feature request] Add option to select the right USB device.

ChrisAnd1998, 2021-03-09:
Related to my previous issue: https://github.com/zeevy/grblcontroller/issues/135
Having an option to select the right USB device will fix my issue. The app doesn't connect to GRBL because it's automatically picking the wrong USB device :/

[image]

ChrisAnd1998, 2021-03-10:
@zeevy I can manually fix this my hardcoding line **291** in **GrblUsbSerialService.java** to 
 `if (deviceVID == 0x2341 && devicePID == 0x0042)  {`
Now it keeps asking for the right device.. However i'm unable to build a proper APK. It keeps crashing on connecting to a device. Has this something to do with running the apk in debug mode?

zeevy, 2021-03-12:
Thanks for the update, I will check this in future releases