zeevy/grblcontroller #137: Upgrade promt cancels nc-program

itziweb, 2021-03-15:
Hello!

At irregular intervals, the message appears asking for the update to the + version of the software.
It interrupts an running program and it cannot be resumed which is a little bit anoying.

I want to support this great development anyway - so is there any other way besides the google play store to purchase the plus version?

Kind regards!

zeevy, 2021-03-15:
I am not sure on irregular intervals, program ask to upgrade only once per launch.

Unfortunately there is no other way you can contribute.
You can get a promo code by sending a mail to app developer in about section of the app, but you need to use that code in google play store

itziweb, 2021-03-15:
Thanks for your response!

Yes, it comes at startup, but it appears as well while the system is in operation and then causing some issues.

As I want no google on the system, i had to install the apk directly. It would be nice to offer the app on alternative appstores like f-droid for privacy fanatics like me. :)

zeevy, 2021-04-16:
No, popups comes while system is in operation, I never put any such code.
Please take the latest version and check

closed 2021-04-29 by zeevy