zeevy/grblcontroller #139: Manual mode, continuous movement of f value unstable machine jitter. Step increase, release the key machine can not stop immediately, always hit the machine. Can you improve?

txg5230, 2021-04-05:


zeevy, 2021-04-05:
Releasing the button will not stop the machine. Please use the stop button provided.
Use small step values when you press and holds the button

txg5230, 2021-04-05:
Let go and stop immediately, can you? My machine stroke x1300mm, y2500mm. When moving at high speed, people's reaction is not as fast as the machine.

zeevy, 2021-04-05:
No app is not designed for that, you can enable soft limits in the grbl which can take care of this. Or use very small jog steps. I found this method is most useful in many scenarios

txg5230, 2021-04-05:
Let go and stop immediately, without pressing the stop button, can the app do it

zeevy, 2021-04-05:
No, i am not interested in doing this in app. I feel current implementation is better than this

closed 2021-04-15 by zeevy