zeevy/grblcontroller #142: Grbl hal connection

arlimbad222, 2021-05-26:
Hi @zeevy @terjeio

I am switching grbl to grbl hal is it possible to connect grbl hal with this app i am trying with bluetooth but it's not working but with ugs i can connect arduino due with grbl hal .

https://github.com/terjeio/grblHAL

zeevy, 2021-05-26:
It should work with correct baud rate set. And statup string should be same grbl

arlimbad222, 2021-05-26:
Hi @zeevy

Baud rate is set correctly but what should i set to startup string can you please explain i cant understand what is the meaning of startup string.

Thank you for quick response.

terjeio, 2021-05-26:
See [this wiki page](https://github.com/grblHAL/core/wiki/For-sender-developers) for info.

Grbl for 8-bit Arduino controllers run on hardware that resets the processor on connect, this causes Grbl to send the startup message (or string) automatically. Many controllers that grblHAL runs on does not perform this reset, and often it is not possible to detect a connection either. For these a sender will wait forever for a response that will never come unless they query the controller to detect its presence.

Then there is the startup message itself, grblHAL has many new features and by default sends a sligthly different message than vanilla Grbl so that senders can adjust, some senders do not recognize this. This can be changed by setting a [compatibility level](https://github.com/grblHAL/core/wiki/Compatibility-level) before compiling.

arlimbad222, 2021-05-26:
Hi @terjeio @zeevy 

So as i understand i have to change compatibility level for this controller is that right???

terjeio, 2021-05-26:
> So as i understand i have to change compatibility level for this controller is that right???

It does not hurt to try. Glancing at issues for this sender it seems that it is possible to configure the message so that `grblHAL` is accepted, this may also do the trick. But compiling grblHAL with compatibility level > 0 might be a safer alternative.

arlimbad222, 2021-05-26:
Hi @terjeio 

Okay i will try different levels then get back here.

arlimbad222, 2021-05-26:
Hi @zeevy @terjeio 

I try almost every compatibility level but I fail Bluetooth connected but no communication between the app and grbl what should I try next I attach some photos here.

[image]

[image]

terjeio, 2021-05-26:
Do you have a computer or an app you can use to check the Bluetooth connection with. Connect via a terminal window (such as puTTY for Windows) and send commands manually, `$I` for controller info - `?` for real-time status. If these do not respond there is a problem somewhere.

One reason could be that the [wrong serial port](https://github.com/grblHAL/SAM3X8E/blob/master/grblHAL_Due/src/serial.h) is configured in the controller or that the RX/TX pins are not wired correctly (not crossed). The pins you have connected are shared with the USB programming port, don't know if this could be an issue, perhaps try with `#define SERIAL_DEVICE 2` and use those pins instead?

arlimbad222, 2021-05-26:
Hi @zeevy @terjeio 

I try with compatibility level 2 and USB connection for my Android device and it works perfectly but with bluetooth it creates problem i don't know why ????

terjeio, 2021-05-26:
> i don't know why ????

It could be as simple as having the USB cable plugged into the programming port as the pins you have connected the module to (RX/TX) are shared with this. Maybe it works with the USB cable unplugged and an external power supply connected to the barrel jack? Or by switching to serial device 2 which uses these pins:
```
#define SERIAL_RX PIO_PB21A_RXD2
#define SERIAL_TX PIO_PB20A_TXD2
```

zeevy, 2021-05-27:
> Hi @zeevy @terjeio
> 
> I try with compatibility level 2 and USB connection for my Android device and it works perfectly but with bluetooth it creates problem i don't know why ????

App is working with the usb connection?

arlimbad222, 2021-05-27:
Hi @zeevy 

Yes app works perfect with usb otg.

zeevy, 2021-05-27:
Ok, then as @terjeio said problem might be in physical connection of Bluetooth module or in baud rate of the module.
1. Check the physical connection
2. Verify the baud rate of the module with Arduino IDE or with any other tools. (make sure Bluetooth baud rate and firmware baud rate both are same)

arlimbad222, 2021-05-27:
Hi @zeevy @terjeio 

I am trying to give external power to the board then connect via bluetooth but still not working

I have one spare Arduino mega2560 i try my bluetooth module with that and it's working properly now this is not bluetooth module issue.

And i cant find any baud rate setting in grbl hal any clue @terjeio how to set baud rate ?[image]

terjeio, 2021-05-27:
> how to set baud rate ?

You have to edit _grbl/serial.c_ - line 46.

I have checked with my Due and both USB and UART mode can be used at the same time(!) Pin 0 and 1 are connected to the USB <> RS232 chip so they work in parallell, I did not know that. Since USB is comms is working then could it be that your Due is somehow faulty? I have an original Arduino Due, yours is a clone? If you have a voltmeter then check that the voltage on pin 0 and 1 is 3.3V.

arlimbad222, 2021-05-27:
Hi @terjeio 

I am using hc-05 Bluetooth module and it's 3.3v module so electronics is not the problem and i test usb with Android app and it's working properly but somehow bluetooth create issue and i spend hours to solve this but i can't is that possible to use another TX-RX pin for communication.

Thanks for quick response.

[image]

terjeio, 2021-05-27:
The Due driver is [now updated](https://github.com/grblHAL/SAM3X8E) with the fix I mentioned above.

Did you try USB communication when the Bluetooth module is connected and if so did that work?

arlimbad222, 2021-05-27:
Hi @terjeio 

I can not understand what you are saying as i understand i use usb connection for pc and then connect bluetooth and then check it again am i right.

And what you update in code for TX-RX pins and how to change communication pins ?

terjeio, 2021-05-27:
> I can not understand what you are saying as i understand i use usb connection for pc and then connect bluetooth and then check it again am i right.

Yes - USB communications should work when the Bluetooth module is attached as they are connected in parallell to the same processor pins. If USB communication is no longer possible then the Bluetooth module is blocking it and I assume that it cannot work either..

> And what you update in code for TX-RX pins and how to change communication pins ?

The other ports simply did not work at all so the fix is for them. Other ports can be selected by editing [my_machine.h](https://github.com/grblHAL/SAM3X8E/blob/master/grblHAL_Due/src/my_machine.h), there is information in that file for which pins they use.

arlimbad222, 2021-05-27:
Hi @terjeio 

Okay I will change that pins by morning and test it again and get back here

Thank you for quick response

arlimbad222, 2021-05-28:
> The Due driver is [now updated](https://github.com/grblHAL/SAM3X8E) with the fix I mentioned above.
> 
> Did you try USB communication when the Bluetooth module is connected and if so did that work?

HI @terjeio 

I just download this code zip file but when I extract the file grbl folder is empty any solutions?

terjeio, 2021-05-28:
> I just download this code zip file but when I extract the file grbl folder is empty any solutions?

The new repo uses submodules so see [this page](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL) which is also linked to from the driver page.

arlimbad222, 2021-05-29:
Hi @zeevy @terjeio 

I try everything possible from my side and the result is i can read output from bluetooth module hc-05 by using serial bluetooth Android application but whenever i send code to board nothing happens any solution for this.

I am using hc-05 at (115200,1,0) stop bit is 1 is that any problem in bluetooth configuration ?

Android app - https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal

[image]

[image]

zeevy, 2021-05-29:
from the image you shared, it seems app is not recognized the grbl firmware none of the buttons seems active.
Can you post image with console tab selected after bt is connected.

[image]

arlimbad222, 2021-05-29:
Hi @zeevy 

I am not at my office right now i will share it  by morning. but i can connect app with usb and app works perfectly but when it comes to bluetooth it creates some problems.

My thought is if it works with usb so this is not grbl issue.

Thank you for quick response

zeevy, 2021-05-29:
After connection with Bluetooth we should see something in the console tab. 
If nothing is there then there will be two possible reasons I can think of
1. Baud rate problem in the Bluetooth module.
2. Connection problem between BT module and the cnc board

arlimbad222, 2021-05-29:
Hi @zeevy 

Baud rate is not an issue because I have one spare Arduino mega 2560 with grbl mega 5x and it works properly with your app with the same Bluetooth module that's why i am confused about problems.

I think @terjeio help me with this

terjeio, 2021-05-29:
> I try everything possible from my side

So you have tried with all the serial ports available? Note that some cannot be used if in use by the board map. Safest to try is by setting SERIAL_DEVICE to 2 as these pins are not allocated for other purposes by any board map (IIRC).

> I am using hc-05 at (115200,1,0) stop bit is 1 is that any problem in bluetooth configuration ?

No.

I have tested all ports with the latest code, but with a FTDI breakout module instead of a Bluetooth module as I do not have any, Anyway, they should behave the same as seen from the controller.

arlimbad222, 2021-05-30:
Hi @zeevy 

I try to send code by Arduino serial monitor 
And connect Bluetooth via grbl controller and i find app dose not connected but it receives deta from board and app console display those codes. So basically I can receive data from the board but can not send data. Any solution for this ???[image]

Hi @terjeio 

I also try to connect Arduino uno via tx rx pin and the result is same as Android app i Arduino can receive data but can not send codes. I attached some pictures hope you will understand what i want to do.[image][image]

zeevy, 2021-05-30:
looks like Bluetooth Tx to CNC board Rx line broken or notworking. 

What is your startup string?

terjeio, 2021-05-30:
First, do **NOT** connect a 5V board (the Arduino Uno) to a 3.3V board (the Due) or you may kill the Due pins or even the whole MCU! I know from experience that it is a bad idea...

Did you ever try the Bluetooth module by switching to serial 2?

Change [my_machine.h](https://github.com/grblHAL/SAM3X8E/blob/master/grblHAL_Due/src/my_machine.h) like this:

//#define USB_SERIAL_CDC     1 // Use native USB port for communication.
//#define SPINDLE_HUANYANG   1 // Set to 1 or 2 for Huanyang VFD spindle. Requires spindle plugin.
//#define KEYPAD_ENABLE      1 // I2C keypad for jogging etc., requires keypad plugin.
//#define EEPROM_ENABLE      1 // I2C EEPROM support. Set to 1 for 24LC16(2K), 2 for larger sizes. Requires eeprom plugin.
//#define EEPROM_IS_FRAM     1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.
#ifndef USB_SERIAL_CDC
**`#define SERIAL_DEVICE      2`**`// Select serial device for output if not using native USB, default is -1, max value is 2`
 #endif
#ifdef SPINDLE_HUANYANG
#define SERIAL2_DEVICE       1 // Select serial device for ModBus output, default is 1, allowed values are 0, 1 and 2
#endif

And connect RX/TX to these pins:

// Serial devices
// -1 (Default)  0 (PA8)  = RX,   1 (PA9)  = TX (Same as USB programming port)
// 0            19 (PA10) = RX,  18 (PA11) = TX
// 1            17 (PA12) = RX,  16 (PA13) = TX
**`// 2            52 (PB21) = RX, A11 (PB21) = TX`**

The startup string when [compatibility level](https://github.com/grblHAL/core/wiki/Compatibility-level) is unset or 0:

_GrblHAL 1.1f ['$' or '$HELP' for help]_

and > 0 (this is exactly the same sa vanilla Grbl):

_Grbl 1.1f ['$' for help]_

One issue to be aware of when connecting via a Bluetooth module is that the controller is not reset on connect, so there is no way the controller can know it should send the startup string. The USB connection uses a virtual RTS (or DTR) line to reset the processor and, as the name implies, the startup string is sent as a part of the startup sequence. Not sending a startup string is similar to how many other ARM controllers behaves, and is discussed [here](https://github.com/grblHAL/core/wiki/For-sender-developers). So, to get the startup message when connecting via Bluetooth a soft reset command has to be issued, `CTRL + X.`

There is also some info about inputs [here](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings), it may be relevant since your board is not connected to a machine.

arlimbad222, 2021-05-30:
Hello @terjeio @zeevy 

I try the same method via ugs and I face that same issue ugs connected to Bluetooth device but not get any communication between board and controller. Usb working perfectly with ugs also.

I have that idea because sometimes i use ugs via bluetooth on mega 2560 and it works properly but now i need more frequency that's why i am switching mega to due.

[image]

[image]

terjeio, 2021-05-30:
The DUE has a reset button, do you receive the startup message if you press that?

The USB programming port shares the same MCU pins for communication as RX/TX on pin 0/1, since USB is working RX/TX on pin 0/1 should work too - the controller code does not (and can not) differentiate between them. So for RX/TX via pin 0/1 you can be 100% sure there is no firmware fault. The RX line has a chip that isolates the USB RX from the pin 0 RX, the TX lines are shorted together. If there are hardware faults in the controller then either the chip used for isolation is broken or the PCB is defective. The microcontroller and the firmware are ok isnce USB is working.

I read a bit about the module, it defaults to [9600 or 38400](https://create.arduino.cc/projecthub/electropeak/getting-started-with-hc-05-bluetooth-module-arduino-e0ca81) baud and the baud rate must be changed by the controller via AT commands? It cannot be changed by the connecting device? If so you could try with changing the baud rate in the controller to 9600 or 38400 to find out. This is harcoded in serial.c to 115200.

If this fails too, do you have acces to an oscilloscope or a logic analyzer?

arlimbad222, 2021-06-01:
Hi @terjeio @zeevy 

I try different baud rate but  I face the same problem at a different baud rate. 

Whenever I push that reset button I receive the startup message yes it's working which means I can receive data from due. i attached some screenshots @zeevy please see those photos I think you can help me out with this.

I ordered a new board today if the board was faulty I will get back here when I receive the new due.

[image]

zeevy, 2021-06-02:
If the BT and usb uses same rx and tx pins then commands sent by the app may not reach the microcontroller. App got no response for the commands $I $$ etc... Which indicates commands are not received by the controller

terjeio, 2021-06-11:
@arlimbad222 FYI I have ordered a couple of BT modules for testing and it is likely I am going to make a grblHAL plugin for configuring. I'll keep you informed.

arlimbad222, 2021-06-11:
Hi @terjeio 

Thank you so much for efforts i order another Arduino due but i receive damaged board so i order it again and now i am out of town for couple of weeks.

If any good news about bluetooth i am very happy to know it.

Thanks again and again.. have a great day.

terjeio, 2021-06-25:
@arlimbad222 I finally got a BT module today and, after configuring, it worked as it should.

I put it into AT mode and sent these command to configure it:

```
AT
AT+UART=115200,1,0
AT+POLAR=1,1
```

I am working on a plugin that should allow configuration of the module from the controller. It also adds an input to the controller to be connected to the STATE pin, this will be used to detect a connection so that the the welcome message can be resent on every connect. For testing I would like to know which board map you plan to use so that I can start with that when implementing this feature.

arlimbad222, 2021-06-25:
Hi@terjeio 

I am using ramps 1.6 board for Bluetooth connection and i am using grbl controller app on Android tablet.

terjeio, 2021-06-28:
@arlimbad222 [Bluetooth plugin](https://github.com/grblHAL/Plugins_Bluetooth) is now ready for testing, open a new discussion or an issue there to continue.

Note that the HAS_IOPORTS define in the [Ramps map](https://github.com/grblHAL/SAM3X8E/blob/master/grblHAL_Due/src/ramps_1.6_map.h) has to be uncommented, I am a bit unsure if this should be the default...

With this I believe this issue can now be closed.

arlimbad222, 2021-06-28:
Hi @terjeio 

Sorry for silly questions but I am not an software guy can you please explain more how to add bluetooth plugin I have to download new files or something because I can not find HAS_IOPORTS in ramps_1.6_map.h can you please share step by step details how can I do it.

Thank you for your time.

terjeio, 2021-06-28:
You have to [download or update](https://github.com/grblHAL/core/wiki/Compiling-GrblHAL) to the [latest version](https://github.com/grblHAL/SAM3X8E).

closed 2026-09-23 by zeevy