zeevy/grblcontroller #143: Regarding grbl cnc controller

RasamsettyPranay2002, 2021-08-19:
I am able to connect with both Bluetooth and usb.but when it is connected the buttons are not pop up and they are not working.and the button of the file button also not working.pls solve the problem.pls replyyyy

zeevy, 2021-08-19:
please post the console tab output

which version of grbl firmware you are using?

RasamsettyPranay2002, 2021-08-19:
I am using version 2.56 which is available in play store.It is connecting to Bluetooth but no buttons are working.do you have an app with including voice command mode.if yes can you pls send me to mail.

RasamsettyPranay2002, 2021-08-19:
I am using hc05 Bluetooth module,Arduino Uno,cnc v3 shield, stepper motors,the version of hc05 Bluetooth module is 3.0

zeevy, 2021-08-19:
Can you tell me which version of grbl firmware is used?

Possible reasons for this problem are

### If you are using Bluetooth connection.
- may be you are not using grbl firmware 1.1
- you have not changed the baudrate of the bluetooth modue to 115200

### If you are using USB connection
- may be you are not using grbl firmware 1.1
- the arduino board is too old and its not running the UNO R3 firmware

RasamsettyPranay2002, 2021-08-19:
I am using version 2.56 and i have given baud rate as 115200 as AT command to hc05.can you share the grbl firmware 1.1 to me. pls solve this problem

RasamsettyPranay2002, 2021-08-19:
[image]

zeevy, 2021-08-19:
[image]

In this screen you can find your grbl firmware version

https://github.com/gnea/grbl from here you can download the recent version of grbl

RasamsettyPranay2002, 2021-08-19:
I am able to connect and send the g code in other apps like G-Code2grbl.but i am unable to connect this.in verbose output it is just showing the commands when it is connected but no buttons are working.can you say how to check the firmware and how to upload it through android.i am unable to open that exe file through android (the link that u have kept).or can you give a youtube video link i.e how to upload grblv1.1 firmware.but with other app it is work properly only.pls rectify this problem as soon as possible bro.

RasamsettyPranay2002, 2021-08-19:
[image]

zeevy, 2021-08-20:
This app only supports grbl firmware 1.1.
It seems you have grbl firmware 0.9
Please check in grbl github page on how to upgrade.

I cant help on this as I am not aware of this process

RasamsettyPranay2002, 2021-08-20:
I have upgraded to version 1.1.it is connecting to Bluetooth and the buttons are working.but the problem is when i have uploaded the g code the stepper motors are working to slow and the no of steps are also less.and it is connecting to usb but buttons are not working.i said i have used gcode 2 grbl app.from that  also when i send gcode the motors are moving very slow and the servo motor is not doing its work(pen up and pen down)

zeevy, 2021-08-20:
May be you need to configure the firmware. You can find the more information here https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration ($100, $101, $102, $110, $111, $112, $120, $121, $122 etc..)

I think you are trying to use this app for plotter not a CNC machine. Not sure grbl firmware supports plotters fully or not.

USB Will not work if you have old Arduino board.

RasamsettyPranay2002, 2021-08-20:
I am using this for cnc writing machine where i have used cnc shield v3.can you keep the new Arduino which is compatible with grbl controller (Amazon or any other link to buy).

RasamsettyPranay2002, 2021-08-20:
https://www.amazon.in/dp/B072LPMR5M/ref=cm_sw_r_cp_apa_glt_fabc_SGAN1DW9506SFEMGJH6N
Can i use this Arduino board

zeevy, 2021-08-21:
You can use any arduino board, for usb connection it should have UNO R3 firmware please check with Sellers.

Have you setup your machine settings correctly, if yes your BT connection should work fine

RasamsettyPranay2002, 2021-09-30:
I have installed 1.1 h grbl firmware.i can able to control the stepper motors but the servo motor is not working.it is not going to pen up and down.what would be the problem??pls reply

zeevy, 2021-10-01:
I think grbl does not support servo, its made for cnc

zeevy, 2022-01-07:
> I have installed 1.1 h grbl firmware.i can able to control the stepper motors but the servo motor is not working.it is not going to pen up and down.what would be the problem??pls reply

You can use custom buttons for the servo up and down functionality

closed 2022-01-07 by zeevy