zeevy/grblcontroller #144: Reg voice control cnc

Janaki-Ramana, 2021-08-19:
Hii bro,i have done a Bluetooth cnc writing machine.can you pls provide me the voice controlled grbl controller app.pls so i can make it as voice controlled with your help.hoping that you will send the app link here or to mail.pls this is a request bro

zeevy, 2021-08-19:
Hi Ramana, this application does not have voice control future. I remember someone has already done this (you also commented on that issue). May be @mosabalrojpany can help you on this

closed 2021-08-19 by zeevy

Janaki-Ramana, 2021-08-19:
Bro you pls contact him@mosabalrojpany and ask to send to you.and you pls share that voice controlled apk file to me.so that i can install it.pls do help.because i cant contact him

zeevy, 2021-08-19:
I too don't have his contact