zeevy/grblcontroller #145: Grlb controller android

Pabloaeron, 2021-10-02:
Hello. Im new here. I just have questions,, about the grbl controller. It says connected, but the button i still un availble. Please help

zeevy, 2021-10-02:
Please check these
- you are using grbl 1.1 firmware
- bluethooth connection baudrate is changed to 115200
- you are using latest ardunio board with Uno R3 firmware

Pabloaeron, 2021-10-02:
Thank you. It is working now. I have more question,
I created a gcode file using the "image to gcode" apps from playstore. But after i upload it to grbl controller and try to run it there is and error pop up and It says "unsupported or invalid gcode command found in block".. can i really use the image to gcode apps or i should use inkscape instead?

zeevy, 2021-10-02:
You can use any gcode generator, but need make sure the gcode is compatiable with grbl.
Most of the softwares will have compatiable settings that you may need to change.

closed 2022-01-07 by zeevy