zeevy/grblcontroller #146: android 11 bluetooth problem

caslor, 2021-11-22:
My setup :
cnc shield 4 
arduino Nano 
Bluetooth HC06
grbl v1.1h

Hi i cant make it work with blue tooth connection.
with otg usb cable the app works fine and i can control my cnc

i have programmed my Hc06 blue tooth for  baud rate of : 115200
When i use the app with blue tooth connection the app connects to the blue tooth but the keypad in the app is not becoming active (remains in grey)
when i press any button in the app  the blue tooth look like receive it (led light flash every time i press a button).
while i am still connected with the app to bluetooth if i connect the arduino with a usb cable to a pc (universal g code sender) then in the android app the keypad become active and in the verbose output screen shows the commands i give from the  pc program 
the key pad in the app even it is now active the cnc dont do anything if i press buttons 

any idea?

zeevy, 2021-11-22:
Problem with bluetooth might be its baud rate is not changed to 115200 (please do verify this)

App will not work when both phone and PC are connected. (commands sent by app may not reach the arduino board)

caslor, 2021-11-22:
Hi, Thanks for the quick reply.
i have change my baud rate in Hc06 to 115200
i checked it also with the at comands : 

AT+VERSION
VERSION:3.0-20170609
OK

AT+UART
+UART:115200,1,0
OK

(i have tried and 115200,0,0   setup and didn;t work )

i used different sketch to change my baud rate than the one you have link inside here as these commands didnt work with my HC06 (maybe because of the version they running)

for me the commands that had to use to change the baud rate were : 
 1) Change speed to 115200:
  AT+UART=115200,0,0

2) Change name:
 AT+NAME=DesiredName

about the cables i have checked them lot of times.. 
hc06  RX  pin is connected to arduino nano TX pin on the shield 
hc06  TX  pin is connected to arduino nano RX pin on the shield 

an other reason that i know that the cables are right is because as i told you when the app is running and blue tooth connection is established , if i connect the arduino to the pc and start the program  i can  read the command lines in the app, so bluetooth transmits data back to mobile...

the problem is that keypad is always in Grey after blue tooth connection

caslor, 2021-11-22:
Update to the problem
As i told in the first post i am using a CNC Shield V4  (For Arduino Nano )

i have a spare CNC Shield V3  (For Arduino UNO ) and so i decide to test the app with this one
with UNO shiled everything work fine ( i used the same HC-06 bluetooth i was using with the nano and the same cables)

so the problem is that the program for some reason is not compatible with this version of cnc - Nano

in order to work the nano shield with GRBL software we change the pins in    `  cpu_map.h  `    file (step pins and direction pins for motors are opposite than UNO shield) 

```
// Define step pulse output pins. NOTE: All step bit pins must be on the same port.
  #define STEP_DDR        DDRD
  #define STEP_PORT       PORTD
  #define X_STEP_BIT      5 // 2  // Uno Digital Pin 2 & NANO Digital Pin 5
  #define Y_STEP_BIT      6 // 3  // Uno Digital Pin 3 & NANO Digital Pin  6
  #define Z_STEP_BIT      7 // 4  // Uno Digital Pin 4  & NANO Digital Pin  7
  #define STEP_MASK       ((1<<X_STEP_BIT)|(1<<Y_STEP_BIT)|(1<<Z_STEP_BIT)) // All step bits

  // Define step direction output pins. NOTE: All direction pins must be on the same port.
  #define DIRECTION_DDR     DDRD
  #define DIRECTION_PORT    PORTD
  #define X_DIRECTION_BIT   2 // 5  // Uno Digital Pin 5  & NANO Digital Pin 2
  #define Y_DIRECTION_BIT   3 // 6  // Uno Digital Pin 6  & NANO Digital Pin 3
  #define Z_DIRECTION_BIT   4 // 7  // Uno Digital Pin 7  & NANO Digital Pin  4
  #define DIRECTION_MASK    ((1<<X_DIRECTION_BIT)|(1<<Y_DIRECTION_BIT)|(1<<Z_DIRECTION_BIT)) // All direction bits
```

But i dont know if that change is so important for the serial port connection with bluetooth

zeevy, 2021-11-26:
bluetooth connection will not work if usb action is active

closed 2023-01-16 by zeevy