zeevy/grblcontroller #147: Wemos D1 R32 ESP32 connection problem

IgorPack, 2021-12-12:
Hello.

I am trying to connect this application to ESRDUINO-32 HW-729 (analogue of VeMos D1 R32 ESP32) via bluetooth. The application does not connect, the buttons are inactive. Other similar applications connect successfully via bluetooth and wifi and work.

My settings:
ESRDUINO-32 НW-729
GRBL Controller v2.56
GRBL 1.3a
SerialToserialBT | Arduino 1.8.16 (ESP32test)
Speed 115200

The question is - has anyone already done this? If yes, share your experience.
  
Thank you

zeevy, 2021-12-12:
Can you please provide more info

- What is the baudrate of the Bluetooth module?
- What is the startup string of your grbl firmware?

Please provide the console tab output when you connected to the bluetooth module via app

RTristani, 2021-12-12:
I have not yet gone the bluetooth route.  I am studying tje instructions tjat are included in Controller +.  That's when I found what looked like a typographical error, but wanted to make sure, so i emailed you.Sent from my Verizon, Samsung Galaxy smartphone

IgorPack, 2021-12-14:
Something strange is happening. In the ESP3D web interface settings, I changed the name of the connection to "btespgrbl" and a connection with the phone and PC appeared. But other problems arose.

1. The wifi module stopped working and, as a result, there is no way to connect to the web application, ie, GRBL_ESP stopped appearing in the list of available wifi points.
2. The ability to connect via usb has disappeared, for example, the GRBL Controller does not see the usb controller.
3. It became impossible to uninstall CH340 - moreover, the module itself is installed / reinstalled, but it is impossible to uninstall it in the launch window, a message appears - "No divice is found!". Although in the task manager it is there and is updated.

As I already wrote, the bluetooth speed is 115200, GRBL1.3a, Arduino IDE is the latest version, the COM port is detected when the controller is connected.

[What is the startup line for your grbl firmware?] - where can I see it?

Any idea how to fix the problem?

zeevy, 2021-12-15:
Grbl startup string "Grbl 1.1f ['$' for help]"

You can find more information here https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

Iam not sure on the problem you are facing

IgorPack, 2021-12-28:
На сегодня приложение находит bluetooth (видимо была проблема с платой). Остаётся вопрос по переключению на USB, не может найти устройства. Есть какие-либо мысли на этот счёт?

zeevy, 2021-12-29:
> Today the application finds bluetooth (apparently there was a problem with the board, or with skillful hands). There remains the question of switching to USB, the application cannot find the devices, I tried different cables, the result is the same. Any thoughts on this?
> 
> I love this app and really want to use it, but something goes wrong all the time. And yet, will there be a version for FluidNC, such as CNC Remote Control?

USB I tested only with Uno R3 firmware that will work fine with arduino clones with R3 firmware.

IgorPack, 2021-12-31:
OK, Zeevy, I get you, I'm really sorry that I can't use your application in my projects.

closed 2022-01-19 by zeevy