zeevy/grblcontroller #148: Problems with FluidNC / esp32

irotier, 2022-01-06:
I'm using an esp32 controlled Plasma-CNC with [FluidNC](https://github.com/bdring/FluidNC). I can connect to it and the consol / manual commands seems to work but all the Buttons keep beeing grey and can't be used?!
Befor i had [grbl_esp32](https://github.com/bdring/Grbl_Esp32) set up and everything worked fine
Do you have any suggestions what the problem could be? Because i really love the possibility to control everything from my phone and it worked like a charm...

zeevy, 2022-01-06:
App works only with grbl 1.1 firmware

irotier, 2022-01-06:
Thats realy sad - now i have to choose between using the old system or another method to control my cnc ... and i really loved your app!
But thanks a lot for your fast reply, saved me from trying impossible things for hours ;)

zeevy, 2022-01-06:
> Thats realy sad - now i have to choose between using the old system or another method to control my cnc ... and i really loved your app! But thanks a lot for your fast reply, saved me from trying impossible things for hours ;)

There is still something you can try 

If you are able to customize the FluidNC firmware all you need to do is change the startup string to `Grbl 1.1f ['$' for help]` 
That might work

irotier, 2022-01-07:
It took some time to understand how to compile the firmware (my first time with visual studio), but your workaround did the trick - so thanks a lot!
But would it be possible, to get a version of your app, wich is not checking the grbl-version? The Firmware is still under construction and i have some doubt to make mistakes while selfcompiling it every time... For sure i would buy the full version.

zeevy, 2022-01-08:
I have the option of supporting grbl compatible firmwares in my to do list.

irotier, 2022-01-08:
This sounds great! Thanks again for your support and i will be patient...

closed 2022-01-08 by irotier