zeevy/grblcontroller #149: Alarm messages

Purtacala, 2022-01-30:
I like much GRBL Controller+. I use it on Samsung Note 4 in conjunction of CNC XPRO V5. 
     1. I have a problem with alarm messages that open at the top of the controller covering "Machine Status Panel" for 2 or 3 minutes. All this time I am not able to use the router because this issue. Would be good to be able to set the displayed time in settings. 
     2. Job resume window would be good to have it available at any time (I mean at the press of a button ) in case I want to start the G code somewhere in the middle of the program.
     Thanks for your hard work

zeevy, 2022-01-31:
There may be some thing wrong with the hardware or the gcode.
Long tost message is up to 3.5 seconds only. You may have lot of alarm messages, they are queued up and showing one after another. You can verify this in the console window where you can see the grbl verbose output. Also please let me know what type of alarm message your are getting.

Purtacala, 2022-01-31:
One alarm. Checked the console window and there is one alarm that is displayed for a couple of minutes. There are no multiple alarms lined up to be displayed. Actually, I have tested several situations and I get similar results. Same behavior on Samsung Galaxy S10 E and Samsung Galaxy S5. Thanks for your time.

zeevy, 2022-01-31:
This happens only if there are series of error messages queued up to show. In app did not specified any time for those messages. 
It uses `Toast.LENGTH_LONG` android default value which is 3.5 seconds as per the document.

Can you please try the free version and see if it has the same behavior

zeevy, 2022-01-31:
> One alarm. Checked the console window and there is one alarm that is displayed for a couple of minutes. There are no multiple alarms lined up to be displayed. Actually, I have tested several situations and I get similar results. Same behavior on Samsung Galaxy S10 E and Samsung Galaxy S5. Alarm displayed is "Jog target exceeds machine travel. Job command has been ignored" This is soft limit alarm. I do not use hard limit switches. All messages with a red background stay for long time on display. Messages with black background are short lived, just a few seconds. Thanks for your time.

If your jog distance very long and you keep the jog button pressed, then app will send continues commands to grbl firmware those gets queued in Rx buffer and get executed later may be this is also a problem for you.

Try pressing only once on the jog button

Purtacala, 2022-02-01:
Hi Zeevy
You are right, I should not be holding the jog button down. I tried the
free version by holding down the jog button, but here the alarm message
disappears after a few seconds. I also noticed this alarm is on black
background. Would be a good idea to make the paid version behave like the
free version.
Anyway, this app is very useful for me, I love it. Thanks very much for
your help and I wish you all the best.

Virgil

zeevy, 2022-02-01:
In free version message gets replaced immediately with new one, but in paid version they gets queued up so any new message will wait until the new one is completed

closed 2022-02-01 by zeevy