zeevy/grblcontroller #150: Jog buttons do not work

dmitriybabintsev, 2022-03-31:
Hello.
I faced a pretty strange issue.. Can you please help me fix it?
When I first installed a GRBL controller it worked good.
But something happened and now Jog buttons do not work at all. I don't see any output in the console window (looks like no command sent when pressing any jog button).

zeevy, 2022-03-31:
Which connection type you are using bluetooth or usb?
Only jog buttons are not working?

dmitriybabintsev, 2022-03-31:
I'm using Bluetooth connection.
Only jog buttons are not working. "Homing" button works on that tab.

dmitriybabintsev, 2022-03-31:
I tried reinstalling the app but with no luck.

dmitriybabintsev, 2022-03-31:
I should see a corresponding command in the console window when press any jog button right?

zeevy, 2022-04-01:
> I should see a corresponding command in the console window when press any jog button right?

Yes, it should show the command.
Is there any usb connection also active like phone is connected to any other system via USB cable?

dmitriybabintsev, 2022-04-01:
> > I should see a corresponding command in the console window when press any jog button right?
> 
> Yes, it should show the command. Is there any usb connection also active like phone is connected to any other system via USB cable?

No, only BT connection.

Looks like I found the root cause.

I just noticed that GRBL parameter $10 corrupted in my controller. Its value should be 511, but now $10=1.
All jogging commands work after I set $10=511.

Sorry for triggering a false issue..

closed 2022-04-01 by dmitriybabintsev

zeevy, 2022-04-01:
You're very welcome, thanks for the update