zeevy/grblcontroller #152: No realtime command to get a status report is sent

dmitriybabintsev, 2022-04-06:
Hello.
Can you please advice how to make the grbl controller app periodically send realtime command for requesting status report ('?' or 0x05 or 0x80)?
In the settings I just can see "Status update interval" option and it's set to 150ms.

zeevy, 2022-04-07:
Once it connected to correct hardware, it will actually send the status update command "?" based on the interval selected

dmitriybabintsev, 2022-04-07:
Probably it sends some commands at startup and expects a particular reply?

zeevy, 2022-04-07:
Yes, when app is first connected it will send a rest command and expects the response of Grbl X.Xx ['$' for help] as a startup string

dmitriybabintsev, 2022-04-07:
Got it, thank you. My firmware is sending response "GrblHAL 1.1f ['$' for help]".
When I changed it to "Grbl 1.1f ['$' for help]", the app started sending periodical realtime command.
Are you going to support GrblHAL startup string? Cause by default the latest grbl firmware sends grblHAL on startup:
https://github.com/grblHAL/core/blob/master/report.c
function
void report_init_message (void)

zeevy, 2022-04-08:
App has option to add custom startup string in Settings => Application Settings => Start up String
can you please try adding exact startup string there `GrblHAL 1.1f ['$' for help]`

dmitriybabintsev, 2022-04-08:
Thanks @zeevy , it worked.
May I ask you additional question? I believe this setting (like any other) should be stored somewhere in the config file right?
Can you please tell where is the config file located?

zeevy, 2022-04-09:
Custom startup string is stored in shared preferences of the app.

Check is done here

https://github.com/zeevy/grblcontroller/blob/bea91201d47b88cc568973f1a101234410add38b/app/src/main/java/in/co/gorest/grblcontroller/listeners/SerialCommunicationHandler.java#L118

dmitriybabintsev, 2022-04-12:
The reason I'm asking this is to simplify the process of tuning the application for the customers. To avoid manual configuration by surfing and pressing buttons in GUI, it's much faster just to copy a confgiuration file into some directory.

closed 2022-04-18 by dmitriybabintsev