zeevy/grblcontroller #153: Support for FluidNC (GRBL 3.4)

ricardo-abe, 2022-04-06:
It can connect with FluidNC via bluetooth (https://github.com/bdring/FluidNC) but the UI remains disabled.
The console can be used to send commands and works as expected.
Maybe the app is waiting for a complete set of $$ settings that are missing in FluidNC startup.

zeevy, 2022-04-07:
app needs grbl firmware 1.1 or compatible firmware and needs the startup string same as grbl firmware

ricardo-abe, 2022-04-07:
I see... the startup string is pretty different (as bellow).  Grbl GCode send/response protocol is the same in FluidNC and I'm pretty optimistic that I can continue to use the app on my new ESP32 based controller. I'll try to mimic the startup string somehow or even try to contribute to your project to make it compatible with FluidNC too. 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.4.1]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:cncshield.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:1us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.12]
[MSG:INFO: Axis X (5.000,1005.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.13:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (5.000,1005.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.5:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-995.000,5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.17 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.23:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.19 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.39:low]
[MSG:INFO: BT Started with FluidNC]

Grbl 3.4 [FluidNC v3.4.1 (bt) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

zeevy, 2022-04-07:
I see messages are different than the grbl firmware, app may not work for FluidNC

DirtyEngineer, 2022-04-17:
This would be very helpful to include FluidNC compatibility. It uses all the same gcode and other regular gcode sender's will work with it. Grbl Controller has been very useful and I paid for the premium version. I hope that one day I can use it for controlling FluidNC as I am converting all of my machines to use FluidNC controllers because they are so much more versatile and much faster.

TinkerBoi, 2022-05-24:
+1 Support for FluidNC would be a game changer! Especially for custom built CNC machines.

MeKeCNC, 2022-06-14:
> I see messages are different than the grbl firmware, app may not work for FluidNC

You can provide this support with a few minor changes, you will have a support for those who buy the your software. I bought the premium version also

ricardo-abe, 2022-06-14:
I bought the payed version too and tried to undertand the messages reading the code. 
I'm a C# developer and could help in this change but I don't have the java environment setted in my computer to debug and try to do it.
Whenever I have time, I'll try to contribute.

ricardo-abe, 2022-07-30:
Great news folks.. FluidNC v3.5.0 now supports custom start message.
I have installed and configured the start message as below:
$Start/Message=Grbl 1.1g [\H]

Now Grbl Controller detects the fluidnc and enable all the features properly.
Will connect to the hardware tomorrow and test it with some jobs.

Best Regards.

zeevy, 2022-07-30:
> Great news folks.. FluidNC v3.5.0 now supports custom start message. I have installed and configured the start message as below: $Start/Message=Grbl 1.1g [\H]
>
> Now Grbl Controller detects the [...]

That's a good news, please let us know the working status

ricardo-abe, 2022-08-01:
> That's a good news, please let us know the working status

I have made some tests during the weekend and everything is working perfectly. Jog, Jobs, Homing, Probe.
In my opinion, it's a "must do" upgrade. Complex jobs are now running smoothly. No more gasps when jogging too.

closed 2023-01-16 by zeevy