zeevy/grblcontroller #154: Application send "$10=1" when disconnecting from bluetooth

dmitriybabintsev, 2022-04-18:
Hi @zeevy 
I faced a wierd issue/feature.
When I disconnect from bluetooth module by pressing [image] button, the application always sends "$10=1" command. Next time when I turn it on I will need to manually send "$10=511" command for the application to work correctly.
Can you please tell me what is the need of sending "$10=1" command and how may I avoid this?
[image]

zeevy, 2022-04-19:
Application requires buffer report to work properly.
It sends $10=2 during startup if not already configured in firmware.

On disconnect it will try to restore the original value. As some other applications don't like $10=2
$10 only supports 0 or 1 or 2
Not sure why it needs 511 to work properly.

zeevy, 2022-04-19:
FYI
https://github.com/zeevy/grblcontroller/issues/114

dmitriybabintsev, 2022-04-19:
> Application requires buffer report to work properly. It sends $10=2 during startup if not already configured in firmware.
> 
> On disconnect it will try to restore the original value. As some other applications don't like $10=2 $10 only supports 0 or 1 or 2 Not sure why it needs 511 to work properly.

He @zeevy 
Thanks for your reply.
$10=1 and $10=2 make very poor info in the status messages.
In grblHAL there're lots of options available in $10 parameter:
[image]

If I set $10=1 or $10=2, I will not get info about speed, spindel, position coordinates etc. That's why I need a higher debug level in grblHAL firmware.
And it looks strange for me that the application tries to modify GRBL settings implicitely. Can we somehow avoid this?

dmitriybabintsev, 2022-04-22:
I'm closing the issue in this case, thanks @zeevy

closed 2022-04-22 by dmitriybabintsev