zeevy/grblcontroller #155: grblcontroller connection to grblhal on PI Pico

oke24, 2022-04-20:
I try to connect grblcontroller on Android mobile to a PI Pico with grblhal with RP2040 driver.
The setup as shown with CNCjs/Webbrowser and ioSender/Windows via ser2net on a PI Zero 2 W and PIPico on a PicoBOB works fine.

I tried both Bluetooth and USB connection from the mobile to PI Zero 2W. But I didn't found a way to connect BT or USB to /dev/ttyAMA0. There isn't something similar as ser2net bridge for serial connections via TCP/IP.
I can connect from the mobile to PI Zero with BT, and also from mobile to PI Zero with USB.
But when starting grblcontroller it always says "not connected". 

I would be glad to get some hints how to connect Android mobile, or also if this is not possible. 

[image]

zeevy, 2022-04-21:
Usb may not work as the app supports only few serial chips.
Bluetooth only works if hardware baudrate is matched firmware configured value.

This app is made for grbl firmware 1.1 official. To work with other compatiable firmware you need to provide the startup string in application settings

oke24, 2022-04-24:
Thank you very much for the hint. Modifying the initial GRBL string and connecting with BT and bridging to ttyACM0 with socat solves my problems. My setup is now as shown
[image]

closed 2022-04-24 by oke24