zeevy/grblcontroller #157: Movement Stops in the Middle of a Job

RTristani, 2022-06-06:
LOVE GRBL Control+.  However, lately it has been stopping in the middle of a job.  The spindle will still be spinning, but motion has just "stopped."  Sometimes it will restart when I press the red-box stop, and then press the start/resume buttons.  Other times when I do that, it will ask to restart at last line or fresh start.  Restarting at last line will "scan" through the already-run lines and then send the Z-axis up to a limit switch.  Other times it will resume the job properly.

SainSmart 4030, Samsung Galaxy Tab A 8.0 (2019).

Rob

zeevy, 2022-06-07:
please check these
- usb cable quality in case you use usb connection (EMF issue may cause the app to malfunction)
- power settings in mobile (some mobile android os may not allow the app to run in background for longer time)

RTristani, 2022-06-07:
I will check these.I use a high quality "audio Usb" cable that is supposed to shield it, but I will double check that it isn't cross talking with a motor power cable.Sometimes it works great, and sometimes it stops.  Can never predict it.  I use a usb-C to multiple usb-a converter that also allows me to charge the tablet while it's working on a carving.  I'll check it for crosstalk as well.Sometimes the os allows Grbl Control + to run uninterrupted overnight.  Sometimes it stops after a few hours.  I'll double check USB settings in the tablet.Thank you,RobSent from my Verizon, Samsung Galaxy smartphone

RTristani, 2022-06-10:
No cross talk between the USB cable and any other wiring.

I don’t think it’s the “app running in background” idea because on my last job, it ran for 16 hours straight with no problems.  This time it stopped after 45 minutes.

Any other ideas?

Thank you,

Rob

From: rtristani
Sent: Tuesday, June 7, 2022 10:32 AM
To: zeevy/grblcontroller
Cc: Tristani Robert
Subject: Re: [zeevy/grblcontroller] Movement Stops in the Middle of a Job(Issue #157)

I will check these.

I use a high quality "audio Usb" cable that is supposed to shield it, but I will double check that it isn't cross talking with a motor power cable.

Sometimes it works great, and sometimes it stops.  Can never predict it.  I use a usb-C to multiple usb-a converter that also allows me to charge the tablet while it's working on a carving.  I'll check it for crosstalk as well.

Sometimes the os allows Grbl Control + to run uninterrupted overnight.  Sometimes it stops after a few hours.  I'll double check USB settings in the tablet.

Thank you,

Rob

Sent from my Verizon, Samsung Galaxy smartphone

zeevy, 2022-06-13:
Can you try bluetooth connection?
Usb can be problem in  environment

RTristani, 2022-06-13:
Zeevy - I don't have a bluetooth board on my machine.  I have a very expensive "Audio USB" cable, but on my next carve I will try a shielded printer cable instead.   I'll let you know how it goes.Thank you,RobSent from my Verizon, Samsung Galaxy smartphone

zeevy, 2022-06-13:
May be you can try without stock/material and without the spindle on ( like dry run ). If every thing goes well for few tests then try to run again with spindle on without stock/material

RTristani, 2022-06-13:
Good idea.  I'll try that.Thank you,RobSent from my Verizon, Samsung Galaxy smartphone

closed 2022-07-06 by zeevy