zeevy/grblcontroller #169: Error message, Only GRBL version 1.1 or above is supported.

zalsson, 2023-03-16:
Flashed my arduino uno with  grbl 1.1f then 1.1h but still gets the message "Only GRBL version 1.1 or above is supported" when i connect to my arduino with cnc shield through my android tablet.
Test done with to 2 tablets, 2 arduino, 2 otg adapters same problem.
what am i doing wrong?

zalsson, 2023-03-16:
can it be that i have used some old arduino uno that i had laying around, and not a UNO R3?

zalsson, 2023-03-21:
in console tab, it said. 0.9j!!!! Dowloaded grbl again and wrote it to the arduino and it said 1.1f so now it work.
but i dont understand what went wrong the first couple of times. Because it was the same file i download.
But Hey, iam happy it was that easy to solve.

Thanks @zeevy

zeevy, 2023-03-22:
Great have a nice time with your machine

closed 2023-03-22 by zeevy