zeevy/grblcontroller #171: Android version too new

Brian-Howden, 2023-05-10:
I have a Google Pixel 4A running Android 13.  I went to the playstore to install Grbl Controller and was informed that there were no versions available for my device.  The store complained that the version of Android on my phone was too new.  I went to git hub and sideloaded the most recent version.  It opens on the phone but closes itself within a few seconds.  Are there plans to support newer versions of Android?  Is there another workaround?

zeevy, 2023-05-10:
Its a work in progress. Google brought lot of restrictions in recent version of os and removed support for old code.

Its getting delayed due to lack of time to support this

Brian-Howden, 2023-05-10:
Thanks for the quick update.  The joys of supporting legacy code….

 

Thanks for all the work you have done so far.

 

Brian

xmarduel, 2023-05-18:
Same problem with android 12

LysolPionex, 2023-06-27:
I bet a lot of the users of this app use old android phones to run it as their only use for some old phone, so legacy code is likely, unfortunately, necessary.

LysolPionex, 2023-06-27:
Also, if it helps, you can sideload a copy of the apk onto newer android. I'm able to run it on my 13.1.

Brian-Howden, 2023-06-27:
Interesting.  I did side load a copy onto my Pixel 4a with android 13.x and it opened but immediately crashed.  I gather the author does not feel it should run due to changes in Android.

 

Brian

LysolPionex, 2023-06-27:
Oh, yeah no guarantees it's supported. Mines a oneplus, fwiw.

-=- Brandon

zeevy, 2023-06-27:
It may not work even if we do side load for new android versions.

xmarduel, 2023-06-30:
Hello,
I understand that Zeevy has no time porting the app to newest Android versions.
But I do not understand why he has no time to indicate on this page which version(s) of android is/are suitable.

So Android 12 Google Play does not allow to install grbl-controller. After having downgraded my phone to Android 11, Android 11 said the same thing too. I couldn't find in the samFw page the firmware for my phone (samsung A21s) for android 10.
Nevertheless, I then tried with the help of the app "apkpure" to install GrblController (apkpure has an apk of it), installation succeeded, and I could open the app (yeah!). 
Jogging did work, at least with an USB OTG.

Unfortunately, after connecting my machine back to my computer (with an usb) and using Candle to jog, it worked too, but in Candle the "Work Coordinates" do not change anymore when jogging. Sending a GCode from Candle still work, but there is no feedback concerning the work coordinates. 

What did happened? Which GCode magic commands could fix this ? Thanks in advance for an explanation and how to fix this behavior.

Sincerely,
XM

zeevy, 2023-07-03:
This application requires status report mask value to be 2 for proper working

Tommi55, 2023-12-26:
Just wondering if there is any update on running this system on newer android devices

adoyle-h, 2024-06-14:
What's the progress now? I hope we can use it soon. Thanks.

venkateswararaog-zaggle, 2024-06-14:
You can install the app from release section in github

ho-ho-ho, 2024-06-23:
I'm in the process of trying to get it to work on my phone with Android 13. So far I've encountered 2 big problems: getting permissions to use USB seems to have changed (it straight up crashed as soon as a usb device was connected) and the Material File Picker (for selecting a gcode file) doesn't support the changes in data storage access, showing no files at all. I didn't check if bluetooth has the same permission issues as USB had since I don't have a GRBL(hal) board with bluetooth to test it with.

So far I've managed to get it to run on my phone, but most likely there will be some things I've overlooked so I'd be glad for some feedback: https://github.com/ho-ho-ho/grblcontroller/tree/65da71f6fd937cacfbb23552b0d736f18f9e73e5

zeevy, 2026-09-22:
Version 2.58 is now live on the Play Store. It targets Android 16 (API 36), uses the system file picker for G-code files, and asks for the new Bluetooth and notification permissions. It installs and runs on Android 12, 13 and newer.

One known problem: in USB mode on Android 14 and newer, 2.58 can crash on first launch before a USB device permission is granted. A fix for that is in progress and will go out in the next release. Bluetooth mode is not affected.

If you still see a problem on 2.58, please open a new issue with your phone model and Android version.

closed 2026-09-22 by zeevy