zeevy/grblcontroller #173: Its some suggestions and not issue

mixalhsxalk, 2023-06-06:
Hello,
First of all sorry about my bad English.Im from Greece.Im new to Github and the reason i came here is to give you some ideas for feature updates.I use you software for a while and i thing its the future if you continue to improve it.I use my handmade boards based on Esp32 with Grbl.

Update for Free edition.
1)Nowdays Grbl support BACKLASH option.Its good to have some buttons to control it.
2)Add support for Tablets and Foldable Phones.

Updates for Paid edition.
1)Replace Feed Spindle Rapid Overrrides and Steps Feed with bars.Its a lot easier to use.
2)Please a User Friendly GRBL configurator with calibrator like lightburn or Openbuilds software.
3)Try to insert Probe Tab into Jog Tab.Seems more right and faster to use.
4)A tab with visual live-emulator will be a cool add.

Thank you for your time.Mike.

zeevy, 2023-06-06:
Thank you for your inputs, will check these in upcoming releases. 

`Replace Feed Spindle Rapid Overrrides and Steps Feed with bars.Its a lot easier to use.` this is not possible because in grbl feed change happens at a fixed intervals 

`visual live-emulator ` will take lot of cpu and memory not ideal for mobile devices

mixalhsxalk, 2023-06-06:
Thank you for your fast response.Take a look at FOCUS from sourcerabbit.It has "buttons bar" and its seems pretty  nice and easy to use.Works like button though because of grbl.As for emulation its understandable .I didn't think about it.It can be optional though for the years coming and the constant hardware development.